SatDump/SatDump #503: Minimum elevation not working as expected…

gm8arv, 2023-11-17:
**Description of the issue**

APT data is decoded even when the satellite is below the minimum elevation set.

**Hardware (SDR/PC/OS)**

PC, Windows-11/64.

**Version (Eg, 1.0.0, CI Build 171)**

1.1.2

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

Aang23, 2023-11-17:
Hello!

The intent of minimum elevation was to filter out passes under said elevation at TCA. However, since it's been a common request something to delay the AOS / LOS to a certain elevation will be added soon.

gm8arv, 2023-11-19:
I can see that using TCA also makes sense (Douglas Deans notes this), so at the risk of confusing the user, perhaps either elevation at TCA, or elevation at AOS/LOS would be useful to have (in addition to allowing the TCA users to keep what they have).  Thanks!