SatDump/SatDump #510: Command-line options for automatic processing

gm8arv, 2023-11-18:
**What would you want to be added to satdump?**
1. A command line option for automated processing to display the Recorder panel so that the FFT, Spectrum and Waterfall are visible on startup, e.g. --recorder
2. A command line option for no audio output, e.g. --nosound

**Describe the feature you'd like to have**
As above

**Describe alternatives you've considered**
1. Not aware of any.
2. Can be achieved on Windows with the Volume Mixer.

**Additional context**

Thanks, David

JVital2013, 2023-11-19:
For item 1, perhaps it should default to the recorder screen if `--start_recorder_device` and/or `--engage_autotrack` is specified. Otherwise, I'm thinking this could be a setting instead of a command line option. What are your thoughts?

For item 2, there is now a setting in SatDump UI under > Settings > User Interface called "Play Audio," which can now be turned off and is honored everywhere SatDump may play audio. There's also a button to mute the audio during a decode (which, of course, does not work when using the cli only). Does this meet your needs?

gm8arv, 2023-11-20:
1.  Yes, "--start_recorder_device" should make the Recorder tab the default, and I would suggest that "--engage_autotrack" should (a) open the tracking panel and (b) close the Device and FFT panels.

2. Fine, except that I don't see that setting in my copy (v1.1.2).  When was that added?  Perhaps I need the nightly update?

JVital2013, 2023-11-20:
Makes sense to me. For #2, yes you do need to use a nightly right now to access the option.

gm8arv, 2023-11-21:
Thanks, understood.

gm8arv, 2024-01-04:
Did anyone make any progress on this which I started several weeks back?

"--start_recorder_device" should make the Recorder tab the default, and I would suggest that "--engage_autotrack" should (a) open the tracking panel and (b) close the Device and FFT panels.

Thanks!

JVital2013, 2024-01-06:
You are very generous to call 2 months "several weeks" :-)

Thanks for the patience, this is now implemented and will be a part of the imminent 1.1.3 release.

closed 2024-01-06 by JVital2013

gm8arv, 2024-01-06:
Many thanks, Jamie.  I know what it's like when someone keeps chasing you for things - they might go straight to the bottom of the pile!  Appreciating that it's an amateur effort, I was trying to avoid that.  BTW: will it be in the nightly, now?

JVital2013, 2024-01-06:
Yes, it is in the latest nightly

gm8arv, 2024-01-06:
Perfect!

David
-- 
SatSignal Software - Quality software for you
Web: https://www.satsignal.eu
Email: ***@***.***
Twitter: @gm8arv