SatDump/SatDump #511: Recall form position and size between runs.

gm8arv, 2023-11-18:
**What would you want to be added to satdump?**
The the program remembers its size and position between runs.

**Describe the feature you'd like to have**
The position (top, left) and size (width, height) be remembered between closing the program and its next opening.

**Describe alternatives you've considered**
None.

**Additional context**
Thinking of automated runs, it would be helpful if SatDump appeared at the same position and with the same form size when started, allowing the user to retain a desktop layout.  I have Delphi code for this, but no C/C++.  Category "nice to have".

JVital2013, 2023-11-19:
Great idea - along with this, is the window maximized.

Glfw (SatDump's window backend) has a nice API to get/set this - we just need to work out the best way to load and save it. Right now, the window is created before user preferences are loaded so some initialization rearrangement is needed.

I'll add it to the to do list!

gm8arv, 2023-11-20:
Yes, maximise the window, but only if (a) the window was maximised on closedown or (and my preference) "--max" is specified on the command line.  I don't think it should be automatic as the user may have other things to be shown on the screen.

In my own software (Delphi 2009) I have several events including create, show, and close.  Typically, in the OnCreate I can read the form's saved  location and size, in the OnShow event I can do the command-line parsing and check for "--max" and use those values to set the size and location .  So during your processing of user preferences check for the saved location and size.  I save during the OnClose or OnFree event.

Delighted that you approve of the idea.

ZbychuButItWasTaken, 2023-11-23:
The solution perhaps is not perfect, as the loading screen still appears in the "default" position, however the main UI loads in the last-used location and size.
This should be quite useful. It can be enabled in the settings, it is disabled by default

closed 2023-11-23 by ZbychuButItWasTaken

gm8arv, 2023-11-24:
Many thanks for that, much appreciated.  I hope to test after the weekend (F1/F2 races) so I guess it's in the nightly build by then!

JVital2013, 2023-11-24:
I got it to save maximization too.

I think it's fine for the window to appear in the default location on the splash screen, then move to the correct location once loaded.

Let us know if you have any issues!

gm8arv, 2023-11-25:
Managed to check it early this morning and it works just as stated.  Perfect for my needs.  Thanks!

commit 89c2125e (2023-11-23): option to remember window size and position, fix #511