SatDump/SatDump #512: rtlsdr source not found when in satdump directory

f6bvp, 2023-11-18:
SatDump was downloaded, compiled according to README and installed with sudo make install
Device is  
lsusb :
Bus 003 Device 004: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
From user home directory 
f6bvp@gpredict:~$ satdump sdr_probe
[18:55:10 - 18/11/2023] (E) CPU Does not support AVX2. Extension plugin NOT loading!
[18:55:10 - 18/11/2023] (E) CPU Does not support SSE4_1. Extension plugin NOT loading!
[18:55:11 - 18/11/2023] (I) Found devices (sources) :
[18:55:11 - 18/11/2023] (I) - File Source
[18:55:11 - 18/11/2023] (D)     - type : file
[18:55:11 - 18/11/2023] (D)     - id : 0
[18:55:11 - 18/11/2023] (I) - PlutoSDR IP
[18:55:11 - 18/11/2023] (D)     - type : plutosdr
[18:55:11 - 18/11/2023] (D)     - id : 0
[18:55:11 - 18/11/2023] (I) - Generic RTL2832U OEM #0
[18:55:11 - 18/11/2023] (D)     - type : rtlsdr
[18:55:11 - 18/11/2023] (D)     - id : 0
.....
If cd into satdump directory :
f6bvp@gpredict:~/satdump$ satdump sdr_probe
[19:01:41 - 18/11/2023] (I) Found devices (sources) :
[19:01:41 - 18/11/2023] (I) - File Source
[19:01:41 - 18/11/2023] (D)     - type : file
[19:01:41 - 18/11/2023] (D)     - id : 0
[19:01:41 - 18/11/2023] (I) Found devices (sinks) :
[19:01:41 - 18/11/2023] (I) Done! Goodbye
.....
If cd to / or any other directory satdump sdr_probe is ok !

Regards,

Bernard

JVital2013, 2023-11-19:
I can reproduce the issue - technically, it's trying to load plugins from the `plugins` folder in the satdump repo, and not loading anything since all that's in there is source code.

The intended behavior is to load plugins from the plugins folder in the current working directory, if one exists. So, it is working "as expected," but we might be able to improve it by failing back to system-installed libraries, if applicable, when the plugins folder in the current working directory has no actual plugins.

I'll look at this later.

JVital2013, 2023-11-19:
Fixed as of 10e5a27

closed 2023-11-19 by JVital2013