SatDump/SatDump #515: Baseband Recording Unknown baseband type wav16

f6bvp, 2023-11-19:
https://www.satdump.org/posts/basic-usage/ 
writes 
baseband_format : Same as for offline processing, with a few differences
    wav16 : To record in .wav format (s16, 16-bits)

satdump record :

Found Rafael Micro R820T tuner
[12:25:38 - 19/11/2023] (D) Set RTL-SDR samplerate to 1000000
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
[12:25:39 - 19/11/2023] (D) Set RTL-SDR frequency to -2087926560
[12:25:39 - 19/11/2023] (D) Set RTL-SDR Bias to 0
[12:25:39 - 19/11/2023] (D) Set RTL-SDR AGC to 0
[12:25:39 - 19/11/2023] (D) Set RTL-SDR Gain to 0
[12:25:39 - 19/11/2023] (D) Set RTL-SDR PPM Correction to 0
Allocating 15 zero-copy buffers
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unknown baseband type wav16
./sarsat_live.sh : ligne 12 :  5192 Abandon                 sudo satdump record noaa_baseband noaa_output_directory --source rtlsdr --samplerate 1e6 --baseband_format wav16 --frequency 1544100.3e6 --general_gain 39 --timeout 300

Is this a compilation option compression option ziq ?

73 de Bernard f6bvp

f6bvp, 2023-11-19:
I found the correct format in [README.md](https://github.com/SatDump/SatDump#readme)  

--samplerate [baseband_samplerate] --baseband_format [f32/s16/s8/u8/w16]

JVital2013, 2023-11-19:
Yep, you got it - use w16 instead of wav16. I opened a PR on the site to get that documentation fixed.

For ziq, all you need to do is have libzstd-dev installed. There is also a `-DBUILD_ZIQ=[ON|OFF]` cmake flag, which defaults to "on".

JVital2013, 2023-11-20:
Updated website documentation to be correct. Thanks for the report!

closed 2023-11-20 by JVital2013