SatDump/SatDump #516: satdump live and record display negative frequency

f6bvp, 2023-11-19:
Both live and record options display a negative frequency.

Additionally "satdump record"  does not display command line parameters :

Found Rafael Micro R820T tuner
[13:43:49 - 19/11/2023] (D) Set RTL-SDR samplerate to 1000000
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
[13:43:49 - 19/11/2023] (D) Set RTL-SDR frequency to -2087926560
[13:43:49 - 19/11/2023] (D) Set RTL-SDR Bias to 0
[13:43:49 - 19/11/2023] (D) Set RTL-SDR AGC to 0
[13:43:49 - 19/11/2023] (D) Set RTL-SDR Gain to 0
[13:43:49 - 19/11/2023] (D) Set RTL-SDR PPM Correction to 0
Allocating 15 zero-copy buffers
[13:43:49 - 19/11/2023] (I) Wrote 0 MB
[13:43:49 - 19/11/2023] (I) Wrote 0 MB
[13:43:49 - 19/11/2023] (I) Wrote 0 MB
[13:43:51 - 19/11/2023] (I) Wrote 5 MB
[13:43:51 - 19/11/2023] (I) Wrote 5 MB
[13:43:51 - 19/11/2023] (I) Wrote 6 MB
[13:43:51 - 19/11/2023] (I) Wrote 6 MB
[13:43:51 - 19/11/2023] (I) Wrote 6 MB
[13:43:51 - 19/11/2023] (I) Wrote 7 MB
....

satdump live :

Found Rafael Micro R820T tuner
[13:41:06 - 19/11/2023] (D) Set RTL-SDR samplerate to 2000000
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
[13:41:06 - 19/11/2023] (D) Set RTL-SDR frequency to -2087926560
[13:41:06 - 19/11/2023] (D) Set RTL-SDR Bias to 0
[13:41:06 - 19/11/2023] (D) Set RTL-SDR AGC to 0
[13:41:06 - 19/11/2023] (D) Set RTL-SDR Gain to 0
[13:41:06 - 19/11/2023] (D) Set RTL-SDR PPM Correction to 0
[13:41:06 - 19/11/2023] (D) Parameters :
Allocating 15 zero-copy buffers
[13:41:06 - 19/11/2023] (D)    - agc : 1
[13:41:06 - 19/11/2023] (D)    - baseband_format : "f32"
[13:41:06 - 19/11/2023] (D)    - buffer_size : 1000000
[13:41:06 - 19/11/2023] (D)    - frequency : 1544100300000
[13:41:06 - 19/11/2023] (D)    - samplerate : 2000000
[13:41:06 - 19/11/2023] (D)    - source : "rtlsdr"
[13:41:06 - 19/11/2023] (D)    - start_timestamp : 1700401266.0
[13:41:06 - 19/11/2023] (D)    - symbolrate : 50000.0
[13:41:06 - 19/11/2023] (D)    - timeout : 300
[13:41:06 - 19/11/2023] (I) Module noaa_apt_demod

Could "record mode" display the same in order to check parameters correctness ?

Is --agc 1 the correct syntax ?

73 de Bernard, f6bvp

JVital2013, 2023-11-19:
Can you please share the command you're using to start satdump? It looks like you might be providing the samplerate incorrectly. Satdump live is reporting a requested samplerate of 1544100300000, which is 1.544 THz - somehow, I doubt you have a signal you want to receive here :-)

f6bvp, 2023-11-19:
Yes you are absolutely right ! 
My frequency scientific notation was wrong.
I wanted 1544.1 MHz  --frequency 15441.e5

Sorry for the trouble.

73 de Bernard, f6bvp

closed 2023-11-19 by f6bvp