SatDump/SatDump #517: device permissions usb_open error -3

f6bvp, 2023-11-19:
At first start of satdump live or record I get the following error :

usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
[15:50:01 - 19/11/2023] (E) Fatal error starting device : Could not open RTL-SDR device!

I read  here https://github.com/osmocom/rtl-sdr/tree/master/debian
-------------

Devices are available to users in the plugdev group.

The librtlsdr0 package installs these default rules:
/lib/udev/rules.d/60-librtlsdr0.rules

If you have permissions issues, you may override these values
with your own rules in /etc:

/etc/udev/rules.d/60-librtlsdr0.rules

After editing udev rules, run as root: 
 udevadm control --reload-rules

--------

but I did not succeed.

The found a more simple way to solved the rtl permission issue by adding my user id into /etc/group 
plugdev:x:46:f6bvp

After editing /etc/group I run as root: 
 udevadm control --reload-rules
and disconnect reconnect terminal.

Now satdump live or record is accessing rtlsdr source device.

This could be added somewhere to help.

73 de Bernard, f6bvp

JVital2013, 2023-11-19:
Documenting this is probably a good idea - I'm thinking about working on a "troubleshooting" page that goes over stuff like this, how to troubleshoot the UI if it doesn't load, and more

f6bvp, 2023-11-19:
I was thinking about an FAQ page but a troubleshooting would be an 
excellent idea.

JVital2013, 2023-11-19:
Hm, we might call it FAQ/Troubleshooting to help people find it. Good thought!

JVital2013, 2023-11-28:
We're working on building up documentation over at https://docs.satdump.org, and this one is ready at https://docs.satdump.org/faq_troubleshooting.html.

closed 2023-11-28 by JVital2013