SatDump/SatDump #523: Remote Gui Application Control

benb0jangles, 2023-11-22:
**What would you want to be added to satdump?**
I would like remote GUI application control to be added. 

**Describe the feature you'd like to have**
For example, when using my 3D printer I use Octoprint to control the printer from my computer. The 3D Printer is in garage, Computer is inside house. I use Raspberry Pi with [Octoprint](https://octoprint.org/) to upload, download, control, move axis, so on. It would be cool if Satdump could somehow implement similar system.

**Describe alternatives you've considered**
I consider using [VNC Server](https://ubuntu.com/tutorials/access-remote-desktop#1-overview) with Ubuntu on Raspberry Pi so I can have remote desktop. Then having [folder sync to google drive](https://www.insynchq.com/i/google-drive/linux/ubuntu) using insync. But more complicated & fragmented setup for all users, also remote desktop i think uses lots of cpu of raspberry pi which will be needed for satdump processing perhaps, I don't know exactly because haven't tried yet.

**Additional context**
At present, I use Satdump with my rotator and automatic tracking, i like it very much. However, If I could mount a raspberry pi to the rotator, connect sdr, remote access control GUI web interface, and captured image could be uploaded to cloud drive. This way rotator could be left alone forever. Also adding spyserver type control for global access.

Thanks for reading :)

ZbychuButItWasTaken, 2023-11-23:
Hello!
We have considered a web-based UI for satdump, so it's definitely somewhere up there on the TODO.
I don't remember if we got anywhere further than just the idea, but as far as I am concerned that is coming at some point

Aang23, 2023-11-24:
Well, what's more likely to actually occur is such a system optimized for automated stations meant to be fully headless and have some sort of remote control / monitoring. Doing everything with just the UI won't be as good as something designed for the purpose and would IMO still remain a pretty niche usecase.

Eg, your usecase here could be replaced by a headless service handling everything (like the UI does now). Such a headless operation mode is a much more requested (and IMO - elegant) way to handle it. It also allows not needing any UI stack setup in said usecases (outside of perhaps forwarding some to show a constellation remotely or such).

benb0jangles, 2023-11-25:
But if it is headless, how would one know about rotator errors or failures, rotator calibration issues etc. The system would just carry on regardless with no user feedback. With remote Gui stats, the user could check remotely and view how rotator system is doing which will be important for system refinement especially using diy rotators.

Aang23, 2024-02-16:
With the current headless system there is a Web UI showing rotator status / etc.