SatDump/SatDump #524: Server lost client

f6bvp, 2023-11-22:
Experimenting satdump_sdr_server with goes client example.

Starting satdump_sdr_server on a local console :
...
[16:53:32 - 22/11/2023] (I) Found OpenCL Devices (0) :
[16:53:32 - 22/11/2023] (I) Loading TLEs from /home/bernard/.config/satdump/satdump_tles.txt
[16:53:32 - 22/11/2023] (I) 8854 TLEs loaded!
[16:53:33 - 22/11/2023] (I) Querying SDR Server at 44.168.19.21:5656
[16:53:34 - 22/11/2023] (D) Device File Source
[16:53:34 - 22/11/2023] (D) Device Generic RTL2832U OEM #0
[16:53:34 - 22/11/2023] (D) Device RTL-TCP
[16:53:34 - 22/11/2023] (D) Device SDR++ Server
[16:53:34 - 22/11/2023] (D) Device SpyServer
[16:53:34 - 22/11/2023] (D) Device UDP Source
[16:53:34 - 22/11/2023] (T) FFT Rate 8333, Samplerate 1000000, Final Size 8192, Skip 141
----
On other local console starting :

satdump live goes_grb goes_grb_live_test_client --server_address 0.0.0.0 --server_port 5656 --client
bernard@bernard-f6bvp:~$ satdump_sdr_server 
[16:53:47 - 22/11/2023] (I) Starting SatDump SDR Server v1.1.3-6265d8c1
[16:53:47 - 22/11/2023] (T) Socket successfully bound to TCP port 5656
[16:53:47 - 22/11/2023] (T) Server listening
[16:53:47 - 22/11/2023] (T) Waiting for client!

On satdump-ui selecting Device SRD++ Server, address 0.0.0.0, port 5656, Connect : 

[17:00:11 - 22/11/2023] (I) Starting SatDump SDR Server v1.1.3-6265d8c1
[17:00:11 - 22/11/2023] (T) Socket successfully bound to TCP port 5656
[17:00:11 - 22/11/2023] (T) Server listening
[17:00:11 - 22/11/2023] (T) Waiting for client!
[17:00:15 - 22/11/2023] (T) Server accepted the client
[17:00:15 - 22/11/2023] (T) Waiting for client!
[17:00:15 - 22/11/2023] (D) Ping!

However after a while endless messages :
[17:00:48 - 22/11/2023] (T) Server lost client
[17:00:48 - 22/11/2023] (T) Server lost client
[17:00:48 - 22/11/2023] (T) Server lost client
.....
What am I doing wrong ?
Other question :  I cannot find where to specify sdr device nor bias etc...

73 de Bernard, f6bvp

f6bvp, 2023-11-22:
and in satdump-ui console :

17:19:36 - 22/11/2023] (D) Device UDP Source
[17:19:36 - 22/11/2023] (T) FFT Rate 8333, Samplerate 1000000, Final Size 8192, Skip 141
[17:20:11 - 22/11/2023] (E) Timeout out after asking for UI
[17:20:11 - 22/11/2023] (E) Error connecting to SDR++ Server Timed out

JVital2013, 2023-11-23:
Hello! It sounds like there are 3 operational modes being conflated here, and I'm not quite sure which one you want to use.

- **SDR++ Server** - The SDR++ Server option you're selecting in the SatDump UI is to connect to a SDR++ server, which is a separate piece of software from SatDump. It lets you operate a SDR remotely, but the entire baseband needs to be streamed across the network. I'm guessing this isn't what you want to do - at least not this time!
- **satdump_sdr_server** - SatDump SDR server is our own alternative to SDR++ server. To use it:
  1. Start satdump_sdr_server on a computer on your network.
  2. In SatDump UI on another computer on your network, Click the "refresh" icon next to the SDR list. You should see your remote SDR appear automatically.
  [image]
  3. If automatic discovery isn't working for you, or you need to forward satdump_sdr_server over the internet, you can specify a server in the SatDump UI Settings
  [image]

- **SatDump Client/Server mode for goes_grb/goes_hrit/etc** - This mode is different. Instead of streaming IQ samples across the network, it streams cadu packets between SatDump instances. This way, you can do the first part of the decoding process on one machine attached to a dish, and the rest on another machine, with less bandwidth usage than the options above.

  To do this, you need to run regular-old `satdump` on two computers.
    - On the machine connected to your dish, run satdump like this. You should always have the ip as 0.0.0.0 (except for rare situations), but you can choose another port.
      ```bash
      satdump live goes_grb . --source bladerf --frequency 1686.6e6 --general_gain 28 --dc_block --samplerate 
      14e6 --server_address 0.0.0.0 --server_port 5004
      ```
    - On a remote machine, run satdump like this. Replace the IP address with the IP of the machine you're running satdump in server mode, and update the path.
      ```bash
      satdump live goes_grb /path/to/save/data --server_address 192.168.1.150 --server_port 5004 --client
      ```

Each of these 3 operational modes are independent. I'm not sure which you want to experiment with, but I hopes this helps to clarify things!

f6bvp, 2023-11-23:
Hi Jamie,

You have been very kind to provide such valuable explanations.
You documented very nicely all possibilities with photos.

This let me understand the possible choices and helped me a lot to make it work.

Let me explain what I achieved.

I have a remote site radioclub F4KLO with a patch antenna for SARSAT satellites on 1500 MHz band that works very well. There is a preamplifier inside the patch box and RTL-SDR.COM V3 has biast-T capablility.

There is a fast link between radioclub and my home.
However formerly I did not succeed running satdump-ui through ssh connexion.

Following your excellent explanations I ran satdump_sdr_server on radioclub computer and started satdump-ui at home.
I declared remote SDR settings et voila !

Thank you very much and congratulations for the fabulous soft.
This is just what I needed.

Next step will be recording of SARSAT 2400 bauds telemetry and next step will be to decode BPSK beacons. But this is another challenge.

SARSAT acronyme is for Search And Rescue SATellite. They broadcast individual emergency beacons transmitting on 406 MHz to LUT centers.

73 de Bernard, f6bvp
http://radiotelescope-lavillette.fr/

j.p. : joined image shows on the left Galileo MEOSAR transponder around 1544.1 MHz. Center frequency 1544.5 MHz is for LEOSAR satellites (NOAA, METEOR-M2 & 3, METOP-A, METOP-B,...)
[image]

closed 2023-11-24 by ZbychuButItWasTaken