SatDump/SatDump #525: SDR Bias T not shot down (deactivated) when closing satdump-ui

f6bvp, 2023-11-22:
Using satdump-ui with a RTL-SDR.COM model V3 that has Bias-T capability and it works fine to turn on 5V DC on sma antenna plug for Nooelec preamplifier alimentation.

However if I shutdown satdump-ui without clicking prealably on bias button to stop bias-t it is not switched off (preamplifier LED remains lighted). This means that SDR bias is left activated even without any application working.

I think this should be added in the shutdown part of satdump-ui together with (who kows ?) stopping sdr as if user had also cliked on stop button.

Thanks for such a nice application !

73 de Bernard, f6bvp

f6bvp, 2023-11-22:
Sorry ! shut down would be better than shot down !!!

ZbychuButItWasTaken, 2023-11-23:
hmmm, I don't think there is an easy way to do that, unless we disable the bias-tee every time when stopping the SDR.
@Aang23 what are your thoughts on this?

JVital2013, 2023-11-24:
In my opinion, this makes sense to me - an SDR shouldn't be putting out voltage after it's been told to stop. At least, in my use-cases, this would made sense. The same probably goes for other SDRs with a switchable Bias-T.

It could probably be added into the `stop()` methods.

Aang23, 2023-11-24:
It's definitely doable and a good point. Will be added soon.

commit 159e26d4 (2023-11-24): Shutdown bias tee on shutdown, fixes #525
closed 2023-11-24 by Aang23