SatDump/SatDump #526: Is audio available remotely when using satdump_sdr_server

f6bvp, 2023-11-23:
I read that audio what possibly disabled or enabled when processing NOAA APT FM decoding.
Is audio sound available on satdump-ui client connected to remote satdump_sdr_server ?
Alan already explained me that APT pipeline streaming was not an option.
However could audio AM or FM be provided independently from APT pipeline ?
If not, how could I pipe satdump-ui client stream to local audio ?

Best regards,

Bernard, f6bvp

Aang23, 2023-11-24:
Do you mean with satdump_sdr_server on your remote machine and satdump-ui on a local one? If so, yes it will work as normal.

However if you're running it remotely, say through Anydesk or such, it rather falls upon say remove access SW to forward audio.

f6bvp, 2023-11-25:
Yes I am using : satdump_sdr_server on remote machine and satdump-ui on a local one.
I found Play Audio in Setting on local satdump-ui .
Is is selected but I do not get any audio output.

What do you mean by "remove access SW to forward" ?

Bernard
 

[image]

JVital2013, 2023-11-25:
The "play audio" setting only affects two processing pipelines right now: NOAA APT and Inmarsat Aero. Audio is played when these pipelines are active.

Otherwise, SatDump currently does not have any other support for Audio output.

closed 2023-11-25 by f6bvp