SatDump/SatDump #528: Introduce processing modulation choice (BPSK, QPSK, etc...)

f6bvp, 2023-11-25:
Presently satdump-ui Processing menu offers a choice with a number of satellites.
If by no chance the satellite we are receiving is not in the list how could we process transmitted data ?

I am using remote satdump_sdr_server and local satdump-ui.
Setup is a simple patch 1.5 MHz antenna on which I replaced the preamplifier with a more efficient Nooelec SawBird. A second Nooelec 1.542 MHz Amplifier is attached to RTL-SRD.COM V3 bias-T on PC running satdump_sdr_server. With this setup I am receiving Galileo and GLONASS MEOSAR SARSAT transponders with LNA gain set to only 30 to get same noise level as with AGC. LEOSAR satellites (NOAA, METEOR, METOP) have very strong signals due to low orbits around 800 Km.

My ultimate goal would be to decode SARSAT (Search And Rescue SATellite) telemetry for emergency calls.
LEOSAR transponders retransmit 406 MHz devices emergency beacons. A 2400 bauds beacon is multiplexing the received beacons with other data. 
Satellite transmission centered on 1544.5 MHz is modulated linearly in phase. The Processed Data Stream (PDS) beacon is coded in Biphase NRZ-L at 2400 bauds with a modulation index of 0.39 (SARR-1) and 0.35 (SARR-2).

I would appreciate to have the possibility to select in a sub processing menu not the satellite but rather the modulation type.
This would be a first step toward decoding beacons the format of which is published and available publicly.
https://radiotelescopelavillette.wordpress.com/balises-406-mhz/

73 de Bernard, f6bvp /ai7bg

f6bvp, 2023-11-25:
[image]

Marhmarchi, 2023-11-26:
You can add a json file under the pipelines folder and have one with the parameters you need. For that you need to follow the same formatting as the other pipelines, but changing the options for what you need

Marhmarchi, 2023-11-26:
One thing tho is that currently we don't have a default way to output hard bits, only soft bits, unless you have the sync word/other sync parameters.

Eg. you can't output hard bits from a bpsk signal that you only know the symbol rate of, you need the sync. But, you still have the soft bits.

Marhmarchi, 2023-11-26:
And sarsat support is planned for the future

Marhmarchi, 2023-11-27:
Yes, all are IQ data, the only pipeline that outputs the audio is APT, but separate thing from the recorder.

In regards to ZIQ format, I'm not sure about GR supporting it by default, but, I believe that it is possible to use file meta sink to use IQ data with the ZIQ standard. For that, you'd want to look at `docs/ZIQ.md` for the ZIQ description format. Although I never tried it, from what I saw it is possible to. If you can't, you can still decompress the data and use it as a normal baseband file.

Aang23, 2023-12-02:
I think I should add some tool to convert between baseband formats though.