SatDump/SatDump #534: Option to automatically generate all possible enhancement images

variorex, 2023-11-28:
Hey SatDump Team, I really like the newest additions to the NOAA APT decoder, with the decoding of the telemetry data now possible, the data is now so much richer of information, especially the temperature data, I love it!

When I use the the recorder to automatically track, record  and decode, I normally get the raw image, the separated A and B channel images and also (when the telemetry data is kinda clean of noise) the separated channels 1-5, depending on what the satellite is currently transmitting.
Now, as far as I understand it, the current nightly build now supports only the enhancements (in the viewer tab) that are possible with the data from the available channels, which is a nice step forward, so you don't have to click on the enhancement to only see that they are not compatible with the current dataset.

Now, currently, my workflow after the images are automatically processed and saved consists of going into the viewer tab, enabling the map overlay to show borders and capital cities on the map, and then generating and saving all the enhancement images possible with the available channel data.

**What would you want to be added to satdump?**
An option to enable automatic generation of all enhancements possible with the channels of the currently tracked and processed satellite would be really cool. I think that the offline processor of NOAA APT data does this already (with some enhancements, but not all?), but not the live tracking processor.
Also, maybe an option to automatically generate all these images, but with map overlay, would be really useful.

**Describe the feature you'd like to have**
An option in the settings tab to enable automatic generation of all image enhancements possible with the channels of the currently live tracked satellite and an option to enable generation of a additional version of all these images, but now with additional border and Lat/Lon grid overlay.

I think this not only is relevant to NOAA APT images, but could be an option also useful for all other satellites that produce images where image enhancements are applicable.

**Describe alternatives you've considered**
Doing all these steps manually.

Thanks 😀

RAD750, 2023-11-29:
To avoid clutter, some enhancements have the `autogen` parameter (enables autogeneration of that enhancement) set to `false`. You can override it in your user `settings.json`.

Here is [some documentation](https://docs.satdump.org/composites.html) on the matter. If you still need help, I can fish out the `settings.json` file from my automated station and you can work from there.

JVital2013, 2024-01-15:
Hi @var-IO-reeks,

Do you need any further assistance with this?

var-IO-reeks, 2024-01-16:
Hi @JVital2013,

I got it "partly" working within my automatic tracking + decoding setup, so all I am referring to in the following is related to the CLI version, while the NOAA-APT related stuff described below also goes for my GUI based decoding setup.
I did some messing around within the settings.json file by toggling some of the autogen-entries within the different composite options for the MSU-MR as well as for the AVHRRs as well as adding some new lines of autogen here and there, but with mixed results:

- The NOAA APT decoder always generates the raw.png, the dataset.json, the product.cbor as well as the APT A and B images, when the image had good enough quality, even the AVHRR1-5 channels, the expected behavior. But apart from that, the added autogen-parameters in the json-file don't seem to enable the generation of any of the other possible enhancements like MSA, Enhanced IR, Thermal Channels etc., none of those are generated.

- For LRPT images from Meteor M2-3, the only things my config does generate are the date-coded folder, which always only contains the cadu file, but apart from that none of the images are generated, not even the basic MSU-MR-1-4 images.

If you could provide a json file that does work with these enhancements, that would be really cool, even tho its really not that urgent for me currently, with some exams calling 😉, also haven't investigated that issue that much further to be honest.

But apart from that, the basic automatic tracking station setup works pretty well on a server for about a month now, big thumbs up for that 👍 , I am also looking foreward for a combined CLI decoder and recoder setup that also stores the raw float32 file.

NicolasGagne, 2024-02-03:
I am having the same behaviour on the NOAA ATP side of thing. Same as you I would like to have every enhancement possible.

I have include my settings.json file for your review.
[settings.json](https://github.com/SatDump/SatDump/files/14149450/settings.json)

Hope that help.

NicolasGagne, 2024-02-04:
I have fix it. 
Steps: 
- Uninstall Satdump
- Recompile from source with the following change:
- Adding -DPLUING on the cmake step: " cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr -DPLUGIN_SCRIPTING=ON .."
- Check the output of the cmake step: I was missing the package 'jemalloc' 
- Install it: "sudo apt-get install -y libjemalloc-dev"
- Work as expected now same "settings.json" file.

JVital2013, 2024-02-04:
Hi @var-IO-reeks, @NicolasGagne's example settings.json file is a good example to go off of.

In addition, for anything done in the UI, take a look at your settings in SatDump UI. You want to make sure both **User Interface > Finish Processing when stopping live** and **General SatDump > Automatically Process Products** are turned on. This will effect decoding done in the UI.

For processing done via the command line with `satdump live...`, add the `--finish_processing` flag to ensure all products are created.

For autotrack done via the command line with `satdump autotrack...`, make sure your autotrack.json file has a `finish_processing` line, like this:

[image]

Aang23, 2024-05-14:
This was definitely an instance of missing `finish_processing`. I'll close this for now and re-open if needed.

closed 2024-05-14 by Aang23