SatDump/SatDump #536: Applying CLUT to GK-2A HRIT

eswnl, 2023-11-29:
**What would you want to be added to satdump?**
GK-2A HRIT only outputs one VIS channel and rest IR. Is it possible to automatically apply the CLUT to GK-2A HRIT VIS/IR combination in the same way as for FY-2x?

**Describe the feature you'd like to have**
Output all the normal GK-2A HRIT images + the image with CLUT

**Describe alternatives you've considered**
GeoSatSignal is possible, but is a bit of a hassle. It is annoying trying to get the white balance correct in order to apply the CLUT properly.

**Additional context**
Here you can for provide more information about what you'd like to see as the end result. Feel free to also post images if you have ideas regarding the UI

RAD750, 2023-11-29:
Hi, I'd gladly work on it but living in Europe I'd need to dig a tunnel to receive GK-2A... could you provide the `cadu` file so I can reproduce? 
Thanks!

eswnl, 2023-11-30:
I should be able to get the files. Can you give me your email address so I can send them to you by e.g. web transfer?

eswnl, 2023-12-01:
> John, if you're not free to do this please feel free to @ me and I'll get a cadu sometime over the weekend.
> 
> To add to this request, if we could get the filenames for GK-2A HRIT products to display sequentially (IE move the date/time before the FD_XXX field) it would make post processing animations much easier.
> 
> Here's a screenshot of GK-2A HRIT products sorted by name to highlight what I'm referring to. ![image](https://private-user-images.githubusercontent.com/56950661/287199151-647dc908-952c-4906-8ed3-687baaeb3ed7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE0MjI4MzAsIm5iZiI6MTcwMTQyMjUzMCwicGF0aCI6Ii81Njk1MDY2MS8yODcxOTkxNTEtNjQ3ZGM5MDgtOTUyYy00OTA2LThlZDMtNjg3YmFhZWIzZWQ3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjAxVDA5MjIxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM4YjlmNjljNGY2ZjdjZDFlNWY4NjNlZmRhNDFlMTc0YTk3MTQxMzA3MmMwM2VhMTUwN2Q5OWNmNTNmY2U1OTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.smNmpoJvmDUe-GscWRNZO96A7_uLLHtetgsatRAc9Ak)

Hello Plugger Lockett,

Actually I was going to try and generate some over the weekend, but please feel free to get some yourself and attach them here (it's only cadu file, so it's not large file). Your signal levels are a bit higher than mine, so your setup maybe more reliable for getting them.

I'll try and generate some and keep them just in case.

I suggest mostly fully illuminated disks with slightly visible termination line so we can see the clut in full effect.

eswnl, 2023-12-02:
I created a sample cadu file here:
https://drive.google.com/file/d/17B8Dy_Ytf1kbfsDHCwCKFJDVdC0SLUSh/view?usp=sharing

sergsero, 2023-12-02:
Maybe it will be useful too. The .cadu file from 2023-12-02_03-34_gk2a_hrit_1695Mhz duration of 30 minutes.
https://drive.google.com/file/d/1nG7y6-XM2HXCJFtCW-LRU4wNsaty3jTW/view?usp=drive_link

pluggerlockett, 2023-12-02:
LOL, while both of you were posting my phone was going off with the email notification while I was recording baseband. I'll add those in a bit.

Aang23, 2024-04-25:
@eswnl this should now be quite easy to achieve with the recent large rework of xRIT (and HimawariCast). Do you have the LUT you were using on hand?

eswnl, 2024-04-26:
@Aang23 Yes, I have. But I assumed that SatDump allows you to select the CLUT file. But I don't see that option, but there is a folder "lut" under "resources". I presume I put my CLUT in there.