SatDump/SatDump #537: Bit Depth parameter is not restored as it was selected when saving

f6bvp, 2023-11-29:
When starting satdump-ui almost all saved parameters are correctly restored with previously saved options and values.
However Bit Depth is back to default 8 bit value although 16 bits was set before saving in Settings.

Bernard, f6bvp

JVital2013, 2023-11-29:
Are you referring to when you use the SatDump SDR Server?

f6bvp, 2023-11-29:
Yes. Sorry, this was not obvious.Parameters are restored in two steps. Some as soon as Satdump-ui is started, others when selecting a device source.Bit depth is not.Dr Bernard Pidoux73 de Bernard, F6BVP-- मेरे बहुत-से शौक़ हैंj'ai beaucoup de passe-tempshttp://radiotelescope-lavillette.fr

JVital2013, 2023-12-15:
Sorry for the delayed response on this one - I can reproduce the issue, and we'll have to look at how to fix it. Right now, it's not designed to save the bit depth in the client - it's expecting the server to send the bit depth to use. The problem is, the server never appears to send the bit depth information (let alone send the last used bit-depth). 

To be investigated!

JVital2013, 2023-12-17:
Quick update - we're still working on this - I got a bit distracted with fixing other bugs in the SDR server first 😄. A quick correction on what I last said: settings are saved client-side, and will be loaded from the saved settings soon!

JVital2013, 2023-12-17:
This is now fixed as of the latest commit. Thanks for the report!

commit 04b16663 (2023-12-17): Remote SDR: Load last used bit depth (fixes #537)
closed 2023-12-17 by JVital2013