SatDump/SatDump #539: Issues with FY-4A LRIT and HRIT live decode

pluggerlockett, 2023-12-01:
I rotated my 1.8m dish from GK-2A HRIT over to FY-4A last weekend as it'd been a while since I had checked it. I attempted to get both FY-4A LRIT as well as HRIT. SNR looked good for both services, but no data was decoded, encrypted or in the clear. Historically when I looked at it with my 1.2m offset I was able to get sync and data from both services. LRIT Demodulator looks very strange, and HRIT SNR values are over 2dB higher than my prior attempts that were successful. I attempted decimation of both 2 and 3, and yet I still couldn't get data out. I have a couple cadus from HRIT btw, sadly they weren't end to end as I started the record a little bit after the transmission began. Let me know if you'd like to see them. 

Satdump in use was from the past 2 weeks (I update pretty regularly). Tested on both Debian 12 and Ubuntu 22.04LTS.

FY-4A LRIT
[image]

FY-4A HRIT
[image]

eswnl, 2023-12-01:
I received FY-4A LRIT image the other day. So that's still working.
The constellation looks odd in the LRIT. Looks more like a carrier. Or the data rate has changed.

eswnl, 2023-12-02:
Today's FY-4A LRIT was normal:
[image]

Aang23, 2023-12-02:
Would you be able to provide a baseband that allows reproducing the issue?

pluggerlockett, 2023-12-03:
I'll try and have a look today @Aang23.
I've also swapped to my better/newer laptop. come to find out ants started to make a nest inside of my Thinkpad I had it pointing at GK-2A for so long. Last night I was finally able to get a clean sync and decodes from FY-4A HRIT, with both laptops. But the HRIT live processing seems very finicky. The below screenshot shows some red in the constellation, this will produce no data via the decoder.

[image]

Here's the same transmission after stopping and restarting everything (I also turned the waterfall off to give my Thinkpad a break).

[image]

And a working one from the prior transmission with waterfall after stop/start.

[image]

Finally, here's the one from my newer laptop with the clean decode.

[image]

I don't know, the HRIT service just seems really unstable, at least on my hardware. I should mention that I didn't change anything other than laptops, the rest of my entire setup is static for the duration of these things.

Alan, I took a baseband last night, it was really strange. Looking at it and the HRIT 'hump' it looks close to 12dB SNR, as expected. Offline decode produces numbers in the 2-3dB SNR max. I'll push that up to swisstransfer today and flick it to you.

pluggerlockett, 2023-12-03:
Here's a video I just took of the FY-4A HRIT startup to show some of the issues I'm seeing.

[https://youtu.be/Qu6enGzd7VM](url)

pluggerlockett, 2023-12-03:
And here's the cadu from last night as well.
https://www.swisstransfer.com/d/44988e10-452e-4ae2-a44e-d6d027cebe66

sergsero, 2023-12-03:
In my case, when receiving the signal from FY-4A for a long time, when the signal bandwidth is switched sequentially, there is also the problem of loss of synchronization. I tried to report this in the issues №167 in May:
https://github.com/SatDump/SatDump/issues/167#issuecomment-1537107357

pluggerlockett, 2023-12-04:
> In my case, when receiving the signal from FY-4A for a long time, when the signal bandwidth is switched sequentially, there is also the problem of loss of synchronization. I tried to report this in the issues №167 in May: [#167 (comment)](https://github.com/SatDump/SatDump/issues/167#issuecomment-1537107357)

Just to circle back on this, I believe you're spot on with this comment.

> Interestingly, sometimes the transmission of the image is interrupted in the current narrowband session by switching to broadband. And the missing part of the image is transmitted already in the next time narrowband mode.
Now, due to the forced manual break (stop/start demodulator) of the session, it is not possible to get a whole image.

On this, I was finally able to get a full decode of the Channel 12 image on Saturday, as seen below.

[image]

But what's interesting is that I think the error correction fixed it for me. In the Data Decoder window it appeared I had missed part of the transmission, but when I checked in the Live_Output folder the image was complete.

[image]

eswnl, 2023-12-23:
FY-4B LRIT uses either dual or circular polarisation. I got a higher signal level when using a helical antenna. A benefit is that it helps nullify the GK-2A HRIT signal because it's linear. At the moment, it's transmitting old FY-4A data.
[image]

rbw42, 2024-03-30:
I'm seeing similar partially received image problems as reported here with a new Nooelec antenna/LNA system, looking at the FY-4B LRIT from Perth. The reported SNR typically averages about 9.7, and the constellation plot looks good.

It looks like @pluggerlockett has already provided a bunch of test data, but I'm happy to provide more if that is useful.