SatDump/SatDump #541: SatDump and Himawaricast bbframes not producing images.

eswnl, 2023-12-01:
**Description of the issue**
When I demodulate the signal in SatDump from baseband recording and play back the bbframe in SatDump, images are not being generated. Is it possible the demodulation is not being done correctly/bbframes are corrupted? The modcode displays "5" constantly which is correct I think. The I and Q must be reversed during the demodulation due to the LNB.

**Hardware (SDR/PC/OS)**
AirSpy R2

**Version (Eg, 1.0.0, CI Build 171)**

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
[image]
[image]

Aang23, 2023-12-17:
Could you provide some samples?

eswnl, 2023-12-18:
Got a baseband here:
https://drive.google.com/file/d/1-OKWVbbdHOui2EgyLr8CsYKBcuHwdHwX/view?usp=drive_link

IQ needs to be reversed to play it back. 
For some reason, no cadu/bbframe file is generated.

eswnl, 2023-12-18:
This is what the DVB-S2 lock looks like using the TBS5927.
[image]

eswnl, 2024-01-11:
SatDump is displaying correct modcod = 5 according to this list. But the demodulator doesn't seem to output any file for some reason.
[image]

JVital2013, 2024-02-15:
@eswnl could you delegate access to me for that baseband? I see something that looks amiss on the SatDump side

JVital2013, 2024-02-17:
Could be. The thing I noticed was that the Signal Analyzer program you're using shows a different symbol rate than is in SatDump, but changing it has no effect

eswnl, 2024-02-17:
This is the current pipelines file which I adjusted to new values which may have been the cause.
[image]
Playing back the above recording, SatDump produces the BBFRAME:
[image]

Oh and the recording has to be played back in the "recorder tab" not in the "offline processing". Not sure why this is.
On another recording, looking at the bbframe file, now you can partially see the header.
[image]