SatDump/SatDump #544: Questions about dc_block and dc removal or avoidance

f6bvp, 2023-12-09:
- First : there is no dc_block in satdump-ui menu options. Could it be added ?

- Second : to avoid DC component to be in the recorded passband it is usually suggested to set a different central frequency.
Is using LO parameter together with frequency useful for this ?
I would like to record a SARSAT satellite on 1544,5 MHz using satdump-ui. 
With default LO=0 I set central frequency 1544,250 that what convenient taking into account the Doppler shift that is around +/-30KHz.
However, for further record processing I would like the central frequency to be 1544,5 MHz.
Should I use a combination of LO and frequency parameter for this ?

-third question : 
Doc gives RTL SDR options.

 [SatDump](https://github.com/SatDump/SatDump/tree/master)/[docs](https://github.com/SatDump/SatDump/tree/master/docs)

/SDR-Options.md

Is dc_block a parameters option for satdump record ?
Is it forgotten or not possible to use there ? It is not rejected nor acknowledged when used in command line :

satdump record galileo_baseband galileo_output_directory --source rtlsdr source_id 1 --dc_block --bias  --agc --gain 39 --samplerate 1920e3 --baseband_format w16 --frequency 1544250.e3 --timeout 60

[14:10:30 - 09/12/2023] (D) Device File Source
[14:10:30 - 09/12/2023] (D) Device PlutoSDR IP
[14:10:30 - 09/12/2023] (D) Device Generic RTL2832U OEM #0
[14:10:30 - 09/12/2023] (D) Device Generic RTL2832U OEM #1
[14:10:30 - 09/12/2023] (D) Device RTL-TCP
[14:10:30 - 09/12/2023] (D) Device SDR++ Server
[14:10:30 - 09/12/2023] (D) Device SpyServer
[14:10:30 - 09/12/2023] (D) Device UDP Source
Found Rafael Micro R820T tuner
[14:10:30 - 09/12/2023] (D) Set RTL-SDR samplerate to 1920000
[R82XX] PLL not locked!
[14:10:31 - 09/12/2023] (D) Set RTL-SDR frequency to 1544250000
[14:10:31 - 09/12/2023] (D) Set RTL-SDR Bias to 1
[14:10:31 - 09/12/2023] (D) Set RTL-SDR AGC to 1
[14:10:31 - 09/12/2023] (D) Set RTL-SDR Gain to 39
[14:10:31 - 09/12/2023] (D) Set RTL-SDR PPM Correction to 0
Allocating 15 zero-copy buffers
[14:10:31 - 09/12/2023] (I) Wrote 0 MB
[14:10:31 - 09/12/2023] (I) Wrote 0 MB
[14:10:31 - 09/12/2023] (I) Wrote 1 MB
......

Subsidiary question : I forgot the syntax for choosing second SDR device OEM #1 for satdump record to use it !
I am not sure source_id 1 is correct syntax.
Thanks.

73 de Bernard, f6bvp / ai7bg

JVital2013, 2023-12-15:
1. The DC Block option is available in the UI here as shown below. It's also available under the "processing" section of the recording screen.

    **Note:** DC removal is not reflected in the Recorder FFT/Waterfall display, but it is happening to the samples before being processed.

    [image]

2. To offset the frequency to avoid the DC spike, I find it easier to manually tune to an offset frequency, then use the "Frequency Shift" option in SatDump UI (or `--freq_shift` from the CLI) to tell SatDump where the target signal is in the baseband when decoding. So, in your example, record at 1544.250 MHz, and set the frequency shift setting to 250e3 when decoding to "target" 1544.5 MHz. However, if you don't want to manually set the offset frequency at record time, you can use LO for record, then Frequency Shift for decode.

    [image]

3. Similar to your first question, DC Blocking is only taken into effect when decoding, not recording. As such, DC block is SDR-agnostic and works on recorded basebands as well.

    Subsidiary answer: run `satdump sdr_probe` to get the source_id for your SDR

f6bvp, 2023-12-16:
Thanks a lot for spending your precious time to answer my novice questions.
And TKS for my memory backup about sdr_probe. Now it is written on my notebook.

closed 2023-12-16 by f6bvp