SatDump/SatDump #547: Changing "Avg Rate" to a more user friendly "Number of Averages" in the FFT

db2os, 2023-12-12:
The current implementation of "Avg Rate" is difficult to use.

Typically for DSN operations a high number of averaging is appreciated.
The Minimum setting is 0.010 and the -/+ buttons only step by 0.1
However typically we never use Avg Rate above 0.2 or so.
Thus, 95% of the slider range makes no sense.

Suggest instead of "Avg Rate = 0.01" a "Number of Averages = 100" would make more sense..

**Proposal:
1/Avg Rate = Number of Averages with a range from 1...9999**

Thank you in advance, beautiful Software!
73s de Peter DB2OS

f6bvp, 2023-12-14:
Hi Peter, everyone,
I have just noticed that in today's built of satdump-ui "Avg Num" was working right. Previously it was in the reverse sens or not averaging at all. The only option was to let minimal value 1.
Now it works correctly.
73 de Bernard, f6bvp /ai7bg

db2os, 2023-12-14:
Hi Bernard,
Yes - was surprised this morning with new version..  works great!!  Thanks to the SatDump team!
73s de Peter DB2OS

closed 2023-12-14 by Aang23