SatDump/SatDump #553: HTTP Request Error! and blank Scheduling Window

burgeda, 2023-12-14:
Compiling on Rocky 9.3 was working yesterday after the "autotrack deadlock" issue was resolved however today (2023-12-14) there's a new issue.

Compiles fine but it fails to get any TLE just repeatedly reports that it failed and is retrying.

When "Tracking Configuration " is selected and the window appears the Scheduling tab is just a blank window.

Tried reloading and recompiling several times - same result.

burgeda, 2023-12-14:
Previously resolved issues was #548

burgeda, 2023-12-14:
I've also noticed a change in behaviour when starting it from a comand prompt window by entering 
satdump-ui, it starts up as usual but when I exit by clicking on the x in the upper right corner the gui closes but the cursor never returns to the command prompt window.

When it was working as expected yesterday the cursor returned and the terminal was usuable.

burgeda, 2023-12-14:
Running "ps -ef|grep satdump" in a new terminal shows that /usr/sbin/satdump-ui is still running even if the command prompt that I used to start it is closd.

burgeda, 2023-12-14:
Typo it's /usr/bin/satdump-ui NOT /usr/sbin/satdump-ui

JVital2013, 2023-12-15:
It sounds like either celestrak was down when you were experimenting, or you have been (temporarily) blocked for requesting TLE files too much. The Tracking configuration window will be empty if it does not have any TLEs, which sounds to be the case.

SatDump will repeatedly try to download the files, but it will stop if you are patient enough. This is also why it hangs when exiting: it's still trying to download the TLE files in the background. It should fail out within 2 minutes or so.

Refrain from requesting any data from celestrak for 6 hours, then try again

burgeda, 2023-12-15:
I don't think that's the case as I hadn't requested ANY TLE's for more than 24 hours before I compiled and started it.
While it was failing I cut'n'pasted the TLE it was failing on to my browser and it dowloaded fine.  Several of them did.

Prior versions i.e. yesterdays didn't have a completely black window at this point

Prior versions the cursor reappeared and the terminal window was usable again as soon as satdump was exited.

It's definitely behaving differently than yesterdays code.

---- On Thu, 14 Dec 2023 22:40:37 -0500 Jamie Vital ***@***.***> wrote ---

It sounds like either celestrak was down when you were experimenting, or you have been (temporarily) blocked for requesting TLE files too much. The Tracking configuration window will be empty if it does not have any TLEs, which sounds to be the case.

SatDump will repeatedly try to download the files, but it will stop if you are patient enough. This is also why it hangs when exiting: it's still trying to download the TLE files in the background. It should fail out within 2 minutes or so.

Refrain from requesting any data from celestrak for 6 hours, then try again

JVital2013, 2023-12-15:
Unfortunately, I cannot reproduce the issue. Today's code runs for me, and TLEs load fine. Can you please share your log?

It might be worth completely uninstalling, deleting the contents of your SatDump build folder, and trying again if you have not already.

burgeda, 2023-12-15:
Yesterday I completely reloaded the OS, cloned the git repo, recompiled at least 4 times.

Unfortunately my logs are gone, however there wasn't really anything useful in them other than "HTTP error"

I've blown it away again and I'll reload and repeat the process again later this morning.

Once I've reloaded and if it's still misbehaving I'll upload logs and some screen captures.

---- On Thu, 14 Dec 2023 22:50:58 -0500 Jamie Vital ***@***.***> wrote ---

Unfortunately, I cannot reproduce the issue. Today's code runs for me, and TLEs load fine. Can you please share your log?

It might be worth completely uninstalling, deleting the contents of your SatDump build folder, and trying again if you have not already.

burgeda, 2023-12-15:
Satdump v1.1.3-5d1896c0 compiled and the issues I experienced yesterday related to TLE and tracking aren't happening any longer.  Then again it's not the exact same code.
So I guess in the future I'll wait a day or so after experiencing a hiccup and see if more commits fix things.

This particular issue seems closed, I haven't poked into every nook and crany's yet.  This is AMAZING software and there are a lot of nooks and crany's to experiment with.

For what it's worth I've attached the output of the build process, but since it's working this time it might not be worth anyones time to look at.

[satdump-compile.log.zip](https://github.com/SatDump/SatDump/files/13686733/satdump-compile.log.zip)

Thanks for investigating.

JVital2013, 2023-12-15:
If anything comes up, let us know!

closed 2023-12-15 by JVital2013