SatDump/SatDump #554: Use integer numbers to display satdump-ui FFT Max, Min and Avg Num sliders

f6bvp, 2023-12-15:
There is probably no need to display floating number values on FFT sliders for FFT Max, Min and Avg Num.
Using integer format to display those values would be perfect for dB schedules and average number of FFT.
To me it is rather strange to read that average number of FFT is actually 27.916 for example.
I would be quite satisfied by reading 27 !

Compliments for this fantastic application and continuous progress achieved by developers.

Bernard, 
f6bvp / ai7bg

Aang23, 2023-12-16:
One reason it's done that way is because floats are more efficient internally and used by the backend. Making it int would require int => float conversions everytime.

However the display itself could be set to enforce integers. Thoughts @JVital2013 @RAD750 ?

JVital2013, 2023-12-17:
Yes, changing the format string for the slider should help here. But, I think we should keep one or two decimal places.

The reason is: changes of 0.1 can be visibly seen in the FFT/waterfall, depending on your scale. Additionally, you can change by increments of 0.1 by shift-clicking on the +/- buttons, and increments of 0.01 by ctrl-clicking on them. It makes sense for the slider  number to reflect this specificity.

3 decimal places is definitely more than we need, however. Even two might be a bit more precision than most folks need, but I'd rather not cut it back too far.

f6bvp, 2023-12-25:
Changing average number of FFT with slider hardly select a +1 or -1.
To increase or decrease Avg by one is perfectly ok by clicking + or - labels.
Then there is absolutely no use to display decimals with 3 digits !
You are right to compute all things with simple or double precision floating points but when displaying values on a screen label, again integer values are enough for our eyes.

f6bvp, 2023-12-25:
Presently there is no way to make displayed number appear like say 120.
With the slider you even cannot approach a dedicated value.
And when clicking on + or - you presently only change integer part of it not the decimals that stay the same. 
No use.