SatDump/SatDump #555: Is json "samplerate" parameter only used when Processing ?

f6bvp, 2023-12-16:
samplerate parameters in NOAA.json does not change samplerate on satdump-ui live display.

For example in NOAA.json
    "noaa_hrpt"
...
        "parameters": {
            "samplerate": {
                "value": 3e6
does not change satdump-ui live display
while parameter 
        "frequencies": [
            [
                "NOAA-18",
                1707e6
actually change satdump-ui live frequency display.

I am wrong concluding "samplerate" is for processing only ?

Another questions : where could I find json satdump description ? I cannot find it.
 
And if I edit NOAA.json to add SARSAT frequency it is not taken into effect until I put it in /usr/share/satdump/pipelines/ directory.
When exactly does make process perform copy in that directory ? make or make install does not seem to perform the update !

Aang23, 2023-12-16:
What do you mean exactly? Can you show a screenshot?

As a note, the SARSAT frequency would need to be added on its own, not-yet-existent SARSAT pipeline (though I do want to write a decoder sometime).

JVital2013, 2024-01-01:
Just chiming in here: the samplerate setting in the pipeline does not change the currently selected samplerate for the SDR in the "Recorder" tab. It sets this option, in offline processing:

[image]

JVital2013, 2024-02-04:
I'm looking back through and I realized I did not address two of your questions:

**When exactly does make process perform copy in that directory ? make or make install does not seem to perform the update !**
The pipelines directory is updated with the `make install` command. It updates based on the files in the pipelines folder in the root of the git repository, _not_ from any you may have copied into the build directory.

**Where could I find json satdump description ? I cannot find it.**
Could you please clarify what you're looking for here?

JVital2013, 2024-03-23:
Hi @f6bvp, would you still like assistance with this?

closed 2024-07-30 by JVital2013