SatDump/SatDump #556: Unable to use - doesn't work.

Tomrm96, 2023-12-16:
**Description of the issue**

Tried the nightly build and re-downloading the latest version. When I click device, it detects my RTL-SDR but comes up with a different name ' Generic RTL2832U ' OEM #0 - I do have the 4 not 3 so maybe that's the issue>? 

And nothing happens, where the waterfall goes it just moves it doesn't produce any waveforms. 

When I click on processing, on some of the meteror 2 sats, it just crashes out. All in all it doesn't work. 

**Hardware (SDR/PC/OS)**

Windows 11 -
RTL SDR -4 (Latest and brand new_) 

**Version (Eg, 1.0.0, CI Build 171)**

1.1.3-54d63e1

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

Aang23, 2023-12-16:
Ah... This one indeed. Will be fixed in a bit.

Aang23, 2023-12-16:
When this is done : https://github.com/SatDump/SatDump/actions/runs/7234708788
Try the latest nightly again and it should work with the V4.

Tomrm96, 2023-12-17:
Hey just tested, and it's still not working unfortunately :( 
[image]
[image]

Tomrm96, 2023-12-17:
I should say that I have installed the drivers fine, and it works fine with SDR# and SDR++

Tomrm96, 2023-12-17:
Sorry Had to compress, github only supports 10mb 

https://github.com/SatDump/SatDump/assets/100447956/f6b4c633-d82c-4c91-9c22-9b8afb684e63

Tomrm96, 2023-12-17:
[11:35:51 - 17/12/2023] (E) Usage : C:\Program Files\SatDump\bin\satdump-ui.exe [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[11:35:51 - 17/12/2023] (E) Extra options (examples. Any parameter used in modules can be used here) :
[11:35:51 - 17/12/2023] (E)  --samplerate [baseband_samplerate] --baseband_format [f32/i16/i8/w8] --dc_block --iq_swap
[11:35:51 - 17/12/2023] (D) Starting with OpenGL 4.6.0 NVIDIA 546.33
[11:35:51 - 17/12/2023] (I)    _____       __  ____                      
[11:35:51 - 17/12/2023] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[11:35:51 - 17/12/2023] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[11:35:51 - 17/12/2023] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[11:35:51 - 17/12/2023] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[11:35:51 - 17/12/2023] (I)                                     /_/      
[11:35:51 - 17/12/2023] (I) Starting SatDump v1.1.3-73233fa
[11:35:51 - 17/12/2023] (I) 
[11:35:51 - 17/12/2023] (I) Loading config C:\Program Files\SatDump\share\satdump\satdump_cfg.json
[11:35:51 - 17/12/2023] (I) Loading user settings C:\Users\tomru\AppData\Roaming/satdump/settings.json
[11:35:51 - 17/12/2023] (I) Loading plugins from C:\Program Files\SatDump\lib\satdump\plugins
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\aaronia_sdr_support.dll...
[11:35:51 - 17/12/2023] (E) Error loading C:\Program Files\SatDump\lib\satdump\plugins\aaronia_sdr_support.dll! Error : "C:\Program Files\SatDump\lib\satdump\plugins\aaronia_sdr_support.dll": The specified module could not be found.
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\aim_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin aim_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\airspyhf_sdr_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin airspyhf_sdr_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\airspy_sdr_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin airspy_sdr_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\analog_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin analog_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\bladerf_sdr_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin bladerf_sdr_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\bluewalker3_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin bluewalker3_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\cubesat_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin cubesat_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\dmsp_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin dmsp_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\dscovr_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin dscovr_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\dvb_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin dvb_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\elektro_arktika_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin elektro_arktika_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\eos_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin eos_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\fengyun2_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin fengyun2_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\fengyun3_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin fengyun3_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\fengyun4_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin fenyun4_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\gcom_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin gcom_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\geonetcast_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin geonetcast_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\gk2a_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin gk2a_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\goes_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin goes_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\hackrf_sdr_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin hackrf_sdr_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\himawari_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin himawari_support loaded!
[11:35:51 - 17/12/2023] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\hinode_support.dll...
[11:35:51 - 17/12/2023] (T) Plugin hinode_support loaded!
[... cut ...]
[11:37:50 - 17/12/2023] (I) Demodulating to C:\Users\tomru\OneDrive\Documents/live_output/2023-12-17_11-37_meteor_m2_lrpt_137Mhz/meteor_m2_lrpt.soft
[11:37:50 - 17/12/2023] (I) Buffer size : 163840
[11:37:50 - 17/12/2023] (I) Start processing...
[11:37:50 - 17/12/2023] (I) Using input symbols 
[11:37:50 - 17/12/2023] (I) Decoding to C:\Users\tomru\OneDrive\Documents/live_output/2023-12-17_11-37_meteor_m2_lrpt_137Mhz/meteor_m2_lrpt.cadu
[11:37:50 - 17/12/2023] (I) Progress -nan(ind)%, SNR : 0.000000dB, Peak SNR: 0.000000dB

Aang23, 2023-12-17:
Look under "FFT" and tune Min/Max.

Aang23, 2023-12-17:
Also you should increase the "LNA Gain" on the device, to at least ~40 or so.

Tomrm96, 2023-12-17:
Oh wow I feel stupid now! sorry Im a complete beginner, so hard to find information - thank you! do you have any tips on what the min/max should be? should that stop it crashing on the m2 setting?

Tomrm96, 2023-12-17:
Still getting these, id imigine they are because no signal present?

[image]

Aang23, 2023-12-17:
Yes, since you didn't receive anything between starting and stopping, there was nothing to decode. However what do you mean by crashing?

As for Min/Max (and other settings!), it is honestly more of a personal preference to make the spectrum look as you want. You may prefer something that will show very weak signals well (very "tight" min/max, potentially more Averaging), or more smoothed out with min/max further apart, etc. Usually you do want to make the noisefloor (when there's no signal) "blue" though, and get signals toward yellow/red.

Tomrm96, 2023-12-17:
https://github.com/SatDump/SatDump/assets/100447956/2a0faa9b-1923-49cb-b88a-d2a4a048a953

Its still crashing on this pipeline, and thank you appreciate the help

Aang23, 2023-12-17:
Ah, that's a bug! I'll take a look. However you can still try to receive something, it probably will not crash if you did get some data (I think I know what the bug is).

Tomrm96, 2023-12-17:
Perfect, thank you so much! Appreciate all your work on this :) can't wait to try it out!

closed 2023-12-17 by Tomrm96

Aang23, 2023-12-17:
You're welcome! Feel free to ask anything else if need be.

dennisdebel, 2026-01-28:
I have the same issue on Windows 10, my v4 stick shows up as 'Generic RTL2832U OEM #0'. When I press start, SatDump throws an error, saying "cannot open Generic Generic RTL2832U OEM #0"