SatDump/SatDump #558: NOAA HRPT MHS-Instrument not contrast equalized

LEDFlighter, 2023-12-17:
The MHS-Instrument from NOAA HRPT does not seem to get contrast equalized by default, although it should be.
It was the same like in the HIRS from the DSB a while ago, but this is fixed now.

So it would be nice if this could be corrected.

Aang23, 2023-12-17:
No, the raw channels should not be equalized by default or this prevents using them properly (and not just to "look at". If you want equilazed, adding a composite will be a better idea.

LEDFlighter, 2023-12-17:
Okay, I haven't meant the raw channels, I meant the composites... I understand, that the raw channels should not be equalized, but I think that equalizing the 3 automatically generated composites would be good though.

closed 2024-04-23 by Aang23