SatDump/SatDump #559: Example "autotrack_config.json" file

var-IO-reeks, 2023-12-18:
**What would you want to be added to satdump?**
Hello Satdump Team, some days ago I spotted the WIP "autotrack" option within the CLI version of satdump, but when I tried it out with

`satdump autotrack`

I just got

`(E) Usage : satdump autotrack autotrack_config.json`
`(E) Sample command :`
`(E) ./satdump autotrack autotrack_config.json`

so apparently the usage requires such an autotrack_config.json-file, but after looking through the project files, I failed to find such a file.

**Describe the feature you'd like to have**
Could you add an exemplary autotrack_config.json-file, or add an entry to the wiki explaining how to use the  commands necessary for such a config file, so we can write a custom autotrack config for the satellites we are interested in?

Thanks 😄

Aang23, 2023-12-18:
You can find one there :
https://a-centauri.com/articoli/automating-satellite-reception-with-satdump-cli

But yes, we need to add some on docs.satdump.org perhaps.

var-IO-reeks, 2023-12-18:
Thanks for the link, live processing works well with the instructions given there, but is there an option to simultaneously also save the raw .f32-file, so that the CLI program can fulfill both tasks, just like it is possible with the GUI option, where you can set both the checkmarks "record" and "live" within the options in the "Tracking configuration" window? That would be really useful.

Aang23, 2023-12-18:
It will be coming quite soon yes. As it is WIP I just implemented live processing first.

JVital2013, 2024-03-23:
Recording in autotrack is now working, and an example config file is available at https://docs.satdump.org/autotrack.html.

closed 2024-03-23 by JVital2013