SatDump/SatDump #560: Implementation of built-in (MAX2837) programmable channel filters in HackRF

sergsero, 2023-12-19:
The HackRF One has a MAX2837 chip. It implements programmable channel filters for 1.75 MHz to 28 MHz RF channel bandwidths. 
This allows to remove significant adjacent spectrum interference before the ADC digitizes the baseband.

Probably, channel's bandwidth values greater than 20 MHz (24 and 28 MHz) do not make sense due to the upper sampling limit of 20 MHz. But all from 1.75 to 20 could be useful.

In some programs, this function is implemented, for example, in SDR++. Is it possible to implement this function in SatDump?

JVital2013, 2023-12-21:
Thanks for the suggestion! This is implemented as of 2ed94a28.

[image]

JVital2013, 2023-12-21:
Quick follow-up: all possible filter values have been added, even the ones over 20MHz. In some situations, I believe it is possible to run a HackRF over 20Msps (with sample loss), so it makes sense to include these filters

commit 2ed94a28 (2023-12-21): Implement baseband filters for HackRF (fixes #560)
closed 2023-12-21 by JVital2013