SatDump/SatDump #564: référence indéfinie vers « widgets::FrequencyInput(char const*, unsigned long*, float) »

f6bvp, 2023-12-22:
This error when making satdump happens quite often after git pull.
 
Performing :
Git pull to update satdump
sudo make clean all built files
make
....
[ 39%] Building CXX object src-cli/CMakeFiles/satdump.dir/live.cpp.o
[ 39%] Building CXX object src-cli/CMakeFiles/satdump.dir/main.cpp.o
[ 39%] Building CXX object src-cli/CMakeFiles/satdump.dir/offline.cpp.o
[ 40%] Building CXX object src-cli/CMakeFiles/satdump.dir/record.cpp.o
[ 40%] Building CXX object src-cli/CMakeFiles/satdump.dir/sdr_probe.cpp.o
[ 40%] Building CXX object src-cli/CMakeFiles/satdump.dir/webserver.cpp.o
[ 40%] Linking CXX executable ../satdump
/usr/bin/ld : ../libsatdump_core.so : référence indéfinie vers « widgets::FrequencyInput(char const*, unsigned long*, float) »
collect2: error: ld returned 1 exit status
make[2]: *** [src-cli/CMakeFiles/satdump.dir/build.make:199 : satdump] Erreur 1
make[1]: *** [CMakeFiles/Makefile2:1176 : src-cli/CMakeFiles/satdump.dir/all] Erreur 2
make: *** [Makefile:156 : all] Erreur 2

Back to satdump directory and git pull again 

bernard@bernard-f6bvp:~/satdump/build$ cd ../
bernard@bernard-f6bvp:~/satdump$ git pull
remote: Enumerating objects: 27, done.
remote: Counting objects: 100% (21/21), done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 15 (delta 9), reused 4 (delta 2), pack-reused 0
Dépaquetage des objets: 100% (15/15), 3.96 Kio | 405.00 Kio/s, fait.
Depuis https://github.com/altillimity/satdump
   93709399..4693cc8c  master     -> origin/master
Mise à jour 93709399..4693cc8c
Fast-forward
 resources/scripted_compos/mhs_snowfall.lua | 51 +++++++++++++++++++++++++++++++++++++++++++++++++++
 satdump_cfg.json                           |  7 ++++---
 2 files changed, 55 insertions(+), 3 deletions(-)
 create mode 100644 resources/scripted_compos/mhs_snowfall.lua

Again sudo make clean and make.
Same error.
[ 40%] Building CXX object src-cli/CMakeFiles/satdump.dir/webserver.cpp.o
[ 40%] Linking CXX executable ../satdump
/usr/bin/ld : ../libsatdump_core.so : référence indéfinie vers « widgets::FrequencyInput(char const*, unsigned long*, float) »
collect2: error: ld returned 1 exit status
make[2]: *** [src-cli/CMakeFiles/satdump.dir/build.make:199 : satdump] Erreur 1
make[1]: *** [CMakeFiles/Makefile2:1176 : src-cli/CMakeFiles/satdump.dir/all] Erreur 2
make: *** [Makefile:156 : all] Erreur 2

Then I need to git clone separately a complete new satdump  project to succeed install.

Aang23, 2023-12-22:
Simple! Redo cmake .. :-) !

f6bvp, 2023-12-22:
If I understand well git pull actually overwrites cmake files ?
Maybe I should prevent git to do it ?

Aang23, 2023-12-22:
No, but when we add new files CMake needs to be re-run to detect them. That's it.

f6bvp, 2023-12-23:
Thanks Alan.
I added cmake into cron.daily
...
  cd /home/bernard/satdump/  
  git pull
  cd build
  make clean
  cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr ..
  make
  make install
  date
  echo "mise à jour de satdump effectuée !" 
  date >> /home/bernard/satdump.log
  echo "mise à jour de satdump effectuée !" >> /home/bernard/satdump.log

Aang23, 2023-12-25:
Automated updates, why not!

closed 2023-12-25 by Aang23