SatDump/SatDump #565: I there any project about adding Slow Scan Television (SSTV) from ISS in sadump ?

f6bvp, 2023-12-25:
SSTV is a slow scan TV process sending fixed images coded using audio tones like APT for meteo images.
There is an SSTV transmitter aboard International Space Station ISS already in satdump satellite database (Zarya) ISS.

ISS is quite popular as project ARISS, Amateur Radio on the International Space Station, is a program that lets students experience the excitement of Amateur Radio by talking directly with crew members of the International Space Station. 
[](https://www.ariss.org/)

Basic SSTV description :[]( http://f1ult.free.fr/DIGIMODES/MULTIPSK/sstv.htm)

The mode specifications have been presented on Dayton Paper, JL Barber, "Proposal for SSTV Mode Specifications", 2000:
[](http://www.barberdsp.com/downloads/Dayton%20Paper.pdf)

Usually ISS is sending SSTV images using PD120 mode (126 sec for a color image) or PD180 (187 sec) on 145.800 MHz

2023 Year End Slow Scan TV Event ! from december 27 to december 30, 2023, on 437.800 MHz using PD120 formatting.

73 de Bernard, f6bvp / ai7bg

f6bvp, 2023-12-26:
In order to get ISS SSTV frequencies into satdump to drive SDR  I wrote what we could call a dummy ISS.json derived from NOAA.json that works only for this purpose of frequency settings.
I only kept declarations of noaa_apt_demod and decoder in order to look at what it was like...

By the way satellites Object Numbers look rather located into cryptic file while this is often the best way to identify a satellite. For example ISS (Zarya) NORAD catalog number 25544 only appears commented in satdump_cfg_json like other satellites for satdump is using "NORAD/elements/active.txt" // Most sats".
However object number is not found when processing : 

17:49:45 - 26/12/2023] (I) Saving /home/bernard/satdump/pipelines/2023-12-26_17-49_iss_437Mhz/raw.png...
[17:49:45 - 26/12/2023] (W) Could not get satellite number from parameters or filename!
 
-------------------
{
    "iss": {
        "name": "ISS (Zarya)",
        "live": true,
        "live_cfg": [
            [
                1,
                0
            ],
            [
                2,
                0
            ]
        ],
        "frequencies": [
            [
                "SSTV PD120",
                145.8e6
            ],
            [
                "SSTV PD180",
                437.8e6
            ]
        ],
        "parameters": {
            "samplerate": {
                "value": 1e6
            }
        },
        "work": {
            "baseband": {},
            "audio_wav": {
                "noaa_apt_demod": {
                    "symbolrate": 50e3
                }
            },
            "products": {
                "noaa_apt_decoder": {
                    "audio_samplerate": 50e3
                }
            }
        }            
    }
}

Aang23, 2024-01-02:
While there hasn't been any explicit project started on this side of things at the moment, it's something I've had at the back of my mind for a while. After doing SSTV by going out in the fields for every pass a few years back, I quickly switched to fully automated setups. This has always been more or less of a pain on the decoding side of things, and I always resorted to recording wavs to drop into QSSTV later on.

Hence I've had the idea to add a decoder more suited for this usecase in SatDump for a while... Just need to get to it sometime!

Aang23, 2024-01-02:
Oh and by the way - Happy new year!

f6bvp, 2024-01-03:
Happy new Year !
Lots of success for all your projects investments either professionals or personnals !

closed 2024-01-07 by f6bvp