SatDump/SatDump #567: Unable to launch

Tomrm96, 2023-12-27:
**Description of the issue**
Unable to launch after following install 
**Hardware (SDR/PC/OS)**
Raspberry PI 5, latest Raspberry os 64
**Version (Eg, 1.0.0, CI Build 171)**
followed linux install guide today 
**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
[image]

JVital2013, 2023-12-28:
Is that terminal directly on your Pi's HDMI output, or are you remotely connected over SSH? The UI cannot (easily) be used over SSH

Tomrm96, 2023-12-28:
Ssh mate

JVital2013, 2023-12-28:
For remote use of SatDump UI on a raspberry pi, I'd recommend using a VNC server. If you're using the latest version of Raspberry Pi OS, you can enable it in `raspi-config` and connect to it from your PC using TigerVNC. 

If you need to do it over SSH for some reason, you can - you just need to start the SSH session differently. The framerate may be very low, however. If you're connecting from a Linux box, it can be done without too much trouble; on macOS you must have X11 installed; on Windows you must be running SSH within WSL2 (**not** the original WSL or the built-in Windows SSH command, which is what it looks like you're using). 

1. Connect to your pi with the following ssh command: `ssh -X -Y username@ip`. X and Y are capital, and allow forwarding GUI programs
2. Run satdump-ui like you tried before. This time, it should open up in its own window:
    [image]

Let us know if this helps - we might be able to provide better assistance if you have a specific goal in mind?

Tomrm96, 2023-12-28:
Hi Mate,

Thanks for the help I will give that a go. Ultimately, I'm after setting it up to recieve scheduled photos from the sats.i was looking at NOAAv2 but I haven't managed to get it to play ball with my os and pi 5 and it looks like the project isnt being updated. I use sat dump on my PC, so the ultimate goal is to have my pi 5 take all the photos for me

Tomrm96, 2023-12-28:
I have been taking a look around and I can see that there is some mention of automatic tracking, but I can't seem to find any documentation/ advice on how to do this :( looking in the program under 'schedule and config' I can't seem, to understand how to do it, how would I set it up?

JVital2013, 2023-12-28:
Thanks for the context!

You can use the UI to automate reception - if you go this route, I definately recommend using VNC to run the UI. SSH sessions will close SatDump as soon as you exit them, but programs started over VNC will keep running.

Long-term, if you want something that runs 24/7, your best bet is to use our new command-line based automation, which can be kept running without the UI. Then it can be managed completely over SSH. It's still in beta, so if you choose this route, let us know about any issues you run into!

For a tutorial on using the command-line based automation, check out @RAD750's article here: https://www.a-centauri.com/articoli/automating-satellite-reception-with-satdump-cli.

Tomrm96, 2023-12-28:
Hi, I Appreciate all of your help with this! 

I've followed the guide, but unfortunately it looks like its not starting :( 

[image]

Tomrm96, 2023-12-28:
I'm not quite sure if this would affect it, but I had to use negative values for my ground station:
[image]

Tomrm96, 2023-12-28:
I cleaned my old sat dump of the system, I have downloaded and installed the one suggested: https://www.a-centauri.com/archivio/Radio/satdump_1.1.3-7f0b3505_armhf.deb

However, The actual GUI doesn't load when clicked here: 

[image]

And when trying to run the command I get the following: 
[image]

Tomrm96, 2023-12-28:
Sorry Last message! 

My Mistake, I built It and can confirm the GUI is now running lol 
[image]

But still the same error message 

[image]

JVital2013, 2023-12-28:
Can you attach your full autotrack.json file? It's expecting it to be at /home/username/autotrack.json.

Feel free to set the lat/lon to 0 if you wish to hide your location when sharing

Tomrm96, 2023-12-29:
Hey - thank you for your help again! I put the json in the folder suggested and this worked, however - leaving it to run all night I did not get any results. Nothing saved I ran it a few times, then left it to run until this morning but It gives me the same error message.

[07:31:00 - 29/12/2023] (I) Progress inf%
[07:31:00 - 29/12/2023] (I) Progress nan%
[07:31:10 - 29/12/2023] (I) Progress inf%
[07:31:10 - 29/12/2023] (I) Progress nan%
[07:31:20 - 29/12/2023] (I) Progress inf%
[07:31:20 - 29/12/2023] (I) Progress nan%
[07:31:30 - 29/12/2023] (I) Progress inf%
[07:31:30 - 29/12/2023] (I) Progress nan%
[07:31:40 - 29/12/2023] (I) Progress inf%
[07:31:40 - 29/12/2023] (I) Progress nan%
[07:31:50 - 29/12/2023] (I) Progress inf%
[07:31:50 - 29/12/2023] (I) Progress nan%
[07:32:00 - 29/12/2023] (I) Progress inf%
[07:32:00 - 29/12/2023] (I) Progress nan%
[07:32:10 - 29/12/2023] (I) Progress inf%
[07:32:10 - 29/12/2023] (I) Progress nan%
[07:32:20 - 29/12/2023] (I) Progress inf%
[07:32:20 - 29/12/2023] (I) Progress nan%
[07:32:30 - 29/12/2023] (I) Progress inf%
[07:32:30 - 29/12/2023] (I) Progress nan%
[07:32:40 - 29/12/2023] (I) Progress inf%
[07:32:40 - 29/12/2023] (I) Progress nan%
[07:32:50 - 29/12/2023] (I) Progress inf%
[07:32:50 - 29/12/2023] (I) Progress nan%
[07:33:00 - 29/12/2023] (I) Progress inf%
[07:33:00 - 29/12/2023] (I) Progress nan%
[07:33:10 - 29/12/2023] (I) Progress inf%
[07:33:10 - 29/12/2023] (I) Progress nan%
[07:33:20 - 29/12/2023] (I) Progress inf%
[07:33:20 - 29/12/2023] (I) Progress nan%
[07:33:30 - 29/12/2023] (I) Progress inf%
[07:33:30 - 29/12/2023] (I) Progress nan%
[07:33:40 - 29/12/2023] (I) Progress inf%
[07:33:40 - 29/12/2023] (I) Progress nan%
[07:33:50 - 29/12/2023] (I) Progress inf%
[07:33:50 - 29/12/2023] (I) Progress nan%
[07:34:00 - 29/12/2023] (I) Progress inf%
[07:34:00 - 29/12/2023] (I) Progress nan%
[07:34:10 - 29/12/2023] (I) Progress inf%
[07:34:10 - 29/12/2023] (I) Progress nan%
[07:34:20 - 29/12/2023] (I) Progress inf%
[07:34:20 - 29/12/2023] (I) Progress nan%
[07:34:30 - 29/12/2023] (I) Progress inf%
[07:34:30 - 29/12/2023] (I) Progress nan%
[07:34:40 - 29/12/2023] (I) Progress inf%
[07:34:40 - 29/12/2023] (I) Progress nan%
[07:34:50 - 29/12/2023] (I) Progress inf%
[07:34:50 - 29/12/2023] (I) Progress nan%
[07:35:00 - 29/12/2023] (I) Progress inf%
[07:35:00 - 29/12/2023] (I) Progress nan%
[07:35:10 - 29/12/2023] (I) Progress inf%
[07:35:10 - 29/12/2023] (I) Progress nan%
[07:35:20 - 29/12/2023] (I) Progress inf%
[07:35:20 - 29/12/2023] (I) Progress nan%
[07:35:30 - 29/12/2023] (I) Progress inf%
[07:35:30 - 29/12/2023] (I) Progress nan%
[07:35:40 - 29/12/2023] (I) Progress inf%
[07:35:40 - 29/12/2023] (I) Progress nan%
[07:35:50 - 29/12/2023] (I) Progress inf%
[07:35:50 - 29/12/2023] (I) Progress nan%
[07:36:00 - 29/12/2023] (I) Progress inf%
[07:36:00 - 29/12/2023] (I) Progress nan%
[07:36:10 - 29/12/2023] (I) Progress inf%
[07:36:10 - 29/12/2023] (I) Progress nan%
[07:36:14 - 29/12/2023] (T) Update pass trajectory...
[07:36:14 - 29/12/2023] (C) LOS!!!!!!!!!!!!!! 33591
[07:36:14 - 29/12/2023] (T) Stop pipeline...
[07:36:14 - 29/12/2023] (I) Stop processing
[... 904 lines cut ...]
[09:40:55 - 29/12/2023] (T) Wedge 5 Pos 19 Cor 5599 CAL 20364 19210 21306 21372 20823 20548 20626 20712 CHV 19353 Diff 59023 CH 2 VALID 0
[09:40:55 - 29/12/2023] (T) Wedge 6 Pos 52 Cor 5632 CAL 19268 18327 21696 19632 19637 20594 20528 20838 CHV 20442 Diff 58847 CH 7 VALID 0
[09:40:55 - 29/12/2023] (T) Wedge 7 Pos 87 Cor 5673 CAL 20298 19788 21186 21304 19693 21032 19907 20766 CHV 21313 Diff 57048 CH 4 VALID 0
[09:40:55 - 29/12/2023] (T) Wedge 8 Pos 61 Cor 5588 CAL 19853 19123 20847 20854 20991 20741 21056 20671 CHV 20391 Diff 57504 CH 8 VALID 0
[09:40:55 - 29/12/2023] (T) Wedge 9 Pos 126 Cor 5665 CAL 20533 19143 18746 20610 20482 19677 20339 22090 CHV 20867 Diff 58606 CH 4 VALID 0
[09:40:55 - 29/12/2023] (T) Wedge 10 Pos 62 Cor 5605 CAL 18370 19514 20725 20978 20408 20418 19783 20867 CHV 20480 Diff 56626 CH 6 VALID 0
[09:40:55 - 29/12/2023] (T) Wedge 11 Pos 8 Cor 5662 CAL 19339 20737 21075 20518 19365 20623 20231 21227 CHV 20189 Diff 57460 CH 7 VALID 0
[09:40:55 - 29/12/2023] (C) Couldn't calibrate image with wedges! Likely no valid wedge was identified!
[09:40:55 - 29/12/2023] (I) Saving ./home/admin/Documents/2023-12-29_09-26_noaa_apt_0Mhz/raw.png...
[09:40:55 - 29/12/2023] (I) ----------- Satellite
[09:40:55 - 29/12/2023] (I) NORAD : 25338
[09:40:55 - 29/12/2023] (I) Name  : NOAA-15
[09:40:55 - 29/12/2023] (W) Congratulations for receiving NOAA 15 on APT! It has been 9360 days, 17 hours, 34 minutes and 44 seconds since it has been launched.
[09:40:55 - 29/12/2023] (I) ----------- AVHRR
[09:40:55 - 29/12/2023] (I) Lines : 1699
[09:40:55 - 29/12/2023] (W) Identified channels (debug): A: -1 A1: -1 B: -1
[09:40:55 - 29/12/2023] (I) Saving ./home/admin/Documents/2023-12-29_09-26_noaa_apt_0Mhz/APT-A.png...
[09:40:55 - 29/12/2023] (I) Saving ./home/admin/Documents/2023-12-29_09-26_noaa_apt_0Mhz/APT-B.png...
[09:40:55 - 29/12/2023] (T) Updating elements...
[09:40:55 - 29/12/2023] (T) Update pass trajectory...
^C[10:25:13 - 29/12/2023] (W) Signal Received. Stopping.
[10:25:13 - 29/12/2023] (I) Waiting for the thread...
[10:25:13 - 29/12/2023] (I) Thread stopped
Reattached kernel driver
[10:25:14 - 29/12/2023] (I) Done! Goodbye

Tomrm96, 2023-12-29:
'[09:20:48 - 29/12/2023] (C) Couldn't calibrate image with wedges! Likely no valid wedge was identified!'

nothing is in my documents

pleasefind attached my json without lon and lat 

{
    "parameters": {
		"source": "rtlsdr",
        "samplerate": 1.024e6,
        "gain": 30,
        "bias": true,
        "initial_frequency": 137.5e6       
    },
    "output_folder": "./home/admin/Documents",
    "qth": {
        "lat": 0,
        "lon": 0,
        "alt": 30
    },
    "http_server": "0.0.0.0:8081",
    "tracking": {
        "rotator_config": {
            "rotctl": {
                "address": "127.0.0.1",
                "port": 4533
            }
        },
        "rotator_algo": {
            "park_position": {
                "az": 0.0,
                "el": 90.0
            },
            "park_while_idle": true,
            "unpark_at_minus": 60.0,
            "update_period": 0.1
        },
        "min_elevation": 0
    },
    "tracked_objects": [
        {
            "norad": 25338,
            "frequency": 137.62,
            "live": true,
            "pipeline_name": "noaa_apt",
            "pipeline_params": {
                "autocrop_wedges": false,
                "satellite_number": "15",
                "sdrpp_noise_reduction": true,
                "start_timestamp": -1.0
            }
        },
        {
            "norad": 28654,
            "frequency": 137.9125,
            "live": true,
            "pipeline_name": "noaa_apt",
            "pipeline_params": {
                "autocrop_wedges": false,
                "satellite_number": "18",
                "sdrpp_noise_reduction": true,
                "start_timestamp": -1.0
            }
        },
        {
            "norad": 33591,
            "frequency": 137.1,
            "live": true,
            "pipeline_name": "noaa_apt",
            "pipeline_params": {
                "autocrop_wedges": false,
                "satellite_number": "19",
                "sdrpp_noise_reduction": true,
                "start_timestamp": -1.0
            }
        },
        {
            "norad": 57166,
            "frequency": 137.9,
            "live": true,
            "pipeline_name": "meteor_m2-x_lrpt"
        }
    ]
}

Tomrm96, 2023-12-29:
I've just actually noticed this is different to your screenshot:

mine when running 
[image]

Yours:

[image]

could it be its not on the correct frequency?

JVital2013, 2023-12-29:
Ah, I see what's going on: in your config, you need to add `e6` to the end of all the frequencies, so it will read `137.1e6`, `137.62e6`, and so on.

We'll get the sample file updated; sorry about that!

Tomrm96, 2023-12-29:
Hi Jamie, 

Thanks for your help again.

Unfortunately this hasn't fixed the error, looks to be showing the correct frequency but im still getting the 'c couldn't calibrate wedge' error as I did before changing it 😢 - nothing is saving either as a file, it's like nothing has happened no files/photos in the output folder.

JVital2013, 2024-01-01:
If the frequency display is now correct, but you're not getting any data, there might be something else going on with your setup. The error you're getting is what happens with APT when there is no signal, for one reason or another.

I believe a FFT display is now available on the status webpage; during a pass, do you see any signal on the FFT? If not, it might be worth stopping the autotrack during a pass, and fire up SatDump UI over VNC to see what you can see

closed 2024-01-26 by Tomrm96