SatDump/SatDump #570: [Feature]

777arc, 2023-12-29:
For a USRP device, I spent a while trying to figure out why USRP/UHD wasn't showing up on the list of devices, until I realized I hadn't initialized my B200 yet, and your list of devices is just displaying all active USRPs.  So one idea is when someone has UHD installed, but there are no devices, there could be an item on the list called `(No USRPs found)` or something.  Just an idea.

JVital2013, 2024-01-01:
Thoughts on this @Aang23? I'm not sure it makes sense to say when no SDRs are found. "Silence is golden;" if no USRPs are listed, then none are found.

777arc, 2024-01-01:
The issue is how to communicate "if no USRPs are listed, then none are found" to the user

K4KDR, 2024-01-01:
No idea if this is the best approach or not, but each time I build a new version of SatDump, I copy/paste a specific '**cmake**' command to NOT include support for hardware I don't have:

`cmake -DCMAKE_BUILD_TYPE=Release -DPLUGIN_AIRSPYHF_SDR_SUPPORT=OFF -DPLUGIN_HACKRF_SDR_SUPPORT=OFF -DPLUGIN_LIMESDR_SDR_SUPPORT=OFF -DPLUGIN_PLUTOSDR_SDR_SUPPORT=OFF -DPLUGIN_BLADERF_SDR_SUPPORT=OFF -DPLUGIN_SDRPLAY_SDR_SUPPORT=OFF -DPLUGIN_MIRISDR_SDR_SUPPORT=OFF -DPLUGIN_SPYSERVER_SUPPORT=OFF -DCMAKE_INSTALL_PREFIX=/usr ..`

JVital2013, 2024-01-01:
Yeah, the default is for USRP support to be there, assuming you have the correct libs. Additionally, the log clearly shows which plugins are loaded, including USRP. This seems like UI bloat with very little benefit.

[image]

777arc, 2024-01-01:
Np, was just sharing a point of confusion I ran into with the GUI, I hadn't looked at the logs

JVital2013, 2024-01-01:
Thanks for the suggestion though! I'll let this open for others on the team to look at - maybe they will feel differently. Although, if we do this, we should probably do it for all supported SDRs.

I am but one man with an opinion :-)

777arc, 2024-01-01:
If someone installed satdump with all the SDRs supported then it will clutter the menu, you're right, so this is probably not the best approach

Aang23, 2024-01-02:
Sorry for the late answer, but yes... SatDump is not meant to only support USRPs, and very various SDRs other than USRPs are used. If we only supported a single SDR type this would make sense, but here not finding a specific type is expected on most systems (I am relatively sure I was the only one ever to have every single supported SDR hooked up at the same time as of now...).

As you all pointed out, if we did this for USRPs, we'd want to replicate the behavior for every other SDR making it more confusing and bloated for something which was already covered : If you have built support for a specific device and none shows up, it's not detected (same behavior as SoapySDR or UHD itself).

I'll close the issue, as I think we've established the current behavior is what we should keep.

closed 2024-01-02 by Aang23