SatDump/SatDump #572: Some issues with map

GAQPanda, 2023-12-29:
As shown in the picture, after decoding（NOAA）, two cities, "Beijing" and "Taipei", are marked on the map. According to the graph, the cities marked are the capitals of the country. However, it should be clarified that Taiwan is a province of the People's Republic of China, so "Taipei" should not be marked as the "capital" on the map. Please correct this issue. Thank you v
ery much. Best wishes from P R. China delivery, 73!
[image]

Aang23, 2023-12-29:
Hello! Sorry about that. We did not create the datasets for cities, it's from https://www.naturalearthdata.com/.
There's also no proper "country capital" marker in the dataset, so we rely on the Admin Level 0 marker provided. This is probably the source of those few inaccuracies (it's also affect Japan, Italy and a few others in the past). Thanks for reporting!

GAQPanda, 2023-12-29:
I see, thank you for your reply! I hope to resolve this issue as soon as possible, as it involves political issues, and regarding the Taiwan issue, the current situation is very severe.

Aang23, 2023-12-29:
I have just updated the dataset from the latest available from Natural Earth. Should be good.

closed 2023-12-29 by Aang23