SatDump/SatDump #575: webserver UI not found

Phoenix-64, 2023-12-31:
**Description of the issue**
The cli auto-tracking webui returns a 404 error on 127.0.0.1:8081, the HTML files seem to be missing. 127.0.0.1:8081/api returns the correct data.
Tested on local loop back and externally.  autotracking.json based on:
 https://www.a-centauri.com/articoli/automating-satellite-reception-with-satdump-cli

only modifications for RSPduo and setting http server to 127.0.0.1:8081 and rotor control to 127.0.0.2 to avoid collisions

The nightly build of the 31.12 using the Windows portable cli worked as expected.
https://github.com/SatDump/SatDump/commit/44536312472b62fededcea7cb8409baaace43ec7
**Hardware (SDR/PC/OS)**
RSPduo JetsonNano Ubuntu 20.04.6 LTS

**Version (Eg, 1.0.0, CI Build 171)**
Self build, build arguments: -DBUILD_GUI=OFF
based on commit:
30916bebae5ddc0655227251500fab1cc2699465

JVital2013, 2024-01-01:
Is it possible to share your log, and a screenshot of your web browser with the issue? Note there can't be anything in the URL after the port number, other than a /.

You can try going to 127.0.0.1:8081/status to see if that works as well

Phoenix-64, 2024-01-02:
[image]
[image]
[image]
Here are the screenshots and logs, hope this helps.
Thanks for helping me. 
[satdump-20240102T154008.log](https://github.com/SatDump/SatDump/files/13811004/satdump-20240102T154008.log)

Phoenix-64, 2024-01-06:
I also tested recompiling with gui but that also did not help

JVital2013, 2024-01-17:
Sorry for the delayed response - it's been busy!

**Is it possible you have another service running on 8081 on localhost**? SatDump's 404 page does not look like that - it looks like this page is coming from somewhere else. It might be worth setting `http_server` in autotrack.json to `0.0.0.0:80` (or some other port) to see if that helps.

**Another thing to check:** do you have any browser extensions installed in Firefox? I've seen extensions before that inject tracking information into URLs invisibly. Though, you can usually see tom-foolery like this in the Network section of Developer tools. Try opening Firefox in safe mode (hold shift while opening) to see if that helps.

**Finally, if none of this works**, please share your autotrack config file with us so we can try to reproduce the issue.

Phoenix-64, 2024-01-18:
No problem. I do not think that I have another service running on 8081.
I rebuild with the latest version and on port 80 and 85 but neither worked.
I attached the cmake and make build loges and the autotrack.json file.
Thank you for your help.
[autotrack.json](https://github.com/SatDump/SatDump/files/13980637/autotrack.json)
[cmake_log.txt](https://github.com/SatDump/SatDump/files/13980638/cmake_log.txt)
[make_log.txt](https://github.com/SatDump/SatDump/files/13980639/make_log.txt)

JVital2013, 2024-02-04:
I believe I found the issue: the autotrack config is expecting to have a `fft_enable` option under parameters, but it's missing from your config. Because of this, your config causes unexpected behavior on my end.

I just pushed a fix to handle configs like yours since they should work and there were other related issues too. Can you update and try again? Switch the port back to 8081 as well; processes running as a regular user sometimes won't bind properly to port 80.

Phoenix-64, 2024-02-04:
Thank you for the help. I built it again but it still does not seem to work. I also tried with the fft_enable option but that did not change anything. It's still the same output.

JVital2013, 2024-02-04:
This is a weird one.

Assuming you're back on port 8081, as a sanity check please run `wget -O - http://127.0.0.1:8081/` and see what gets spit out in the terminal.

Phoenix-64, 2024-02-06:
Yea, indeed. Okay I tried fixing some more mist library errors on cmake.
The result of wget is the following:
```
--2024-02-07 00:52:25--  http://127.0.0.1:8081/
Connecting to 127.0.0.1:8081... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-02-07 00:52:25 ERROR 404: Not Found.
```

And without satdump running:
```
--2024-02-07 00:57:07--  http://127.0.0.1:8081/
Connecting to 127.0.0.1:8081... failed: Connection refused.
```

Phoenix-64, 2024-02-07:
I also tried it out on WSL Ubuntu but there I could not get the application to run:
```
(E) Could not connect to Rotcld! Couldn't connect to TCP socket!
terminate called after throwing an instance of 'std::bad_optional_access'
  what():  bad optional access
Aborted
```

JVital2013, 2024-02-07:
Thanks - it sure looks like SatDump is the one returning the 404, since it becomes a "connection refused" when you stop it. What's strange is SatDump never returns a 404 for me - even if you go to a bad page, it returns 200, albeit with page contents of "Error 404".

What version of nng are you using?

Phoenix-64, 2024-02-07:
libnng1
libnng-dev
nng-utils
are all version 1.2.6-1
I tried installing older versions but could not find them with apt. So I built a newer version from the website but got a build error because it is a static and not a dynamic library or vice versa. [nng.nanomsg.org ](nng.nanomsg.org ) Version 1.7.2

Phoenix-64, 2024-02-07:
```
[ 39%] Linking CXX shared library ../libsatdump_core.so
/usr/bin/ld: CMakeFiles/satdump_core.dir/libs/lua/onelua.c.o: in function `os_tmpname':
/home/jetson/SatDump/src-core/libs/lua/loslib.c:172: warning: the use of `tmpnam' is dangerous, better use `mkstemp'
/usr/bin/ld: /usr/local/lib/libnng.a(posix_thread.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `nni_atfork_child' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libnng.a(posix_thread.c.o): in function `nni_plat_init':
posix_thread.c:(.text+0x9b4): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libnng.a(posix_ipcdial.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `ipc_dialer_dial' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libnng.a(posix_ipcdial.c.o): in function `nni_ipc_dialer_alloc':
posix_ipcdial.c:(.text+0x974): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libnng.a(posix_ipclisten.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `ipc_listener_listen' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libnng.a(posix_ipclisten.c.o): in function `nni_ipc_listener_alloc':
posix_ipclisten.c:(.text+0xef0): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libnng.a(posix_resolv_gai.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `resolv_worker' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libnng.a(posix_resolv_gai.c.o): in function `nni_posix_resolv_sysinit':
posix_resolv_gai.c:(.text+0xcb8): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libnng.a(inproc.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `nni_inproc_tran' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libnng.a(inproc.c.o): in function `nni_sp_inproc_register':
inproc.c:(.text+0x12c0): dangerous relocation: unsupported relocation
collect2: error: ld returned 1 exit status
make[2]: *** [src-core/CMakeFiles/satdump_core.dir/build.make:6540: libsatdump_core.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:1339: src-core/CMakeFiles/satdump_core.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

This is the error I get

JVital2013, 2024-02-07:
1.2.6-1 is the same version I have on several of my systems. You can try building nng like this, but I'm not sure it will provide a better result than the apt-based version:

```bash
git clone https://github.com/nanomsg/nng.git
cd nng
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON -DNNG_ENABLE_TLS=ON -DCMAKE_INSTALL_PREFIX=/usr ..
make -j4
sudo make install
```

Any ideas on this issue @Aang23?

Phoenix-64, 2024-04-04:
Okay, I got some time on my hands and tried fixing it again. To no avail, the most recent reales still did not solve the problem. @JVital2013 do you have any recommendations for debugging on Ubuntu?