SatDump/SatDump #576: Error (D) Samplerate sent 250000

f6bvp, 2024-01-01:
While saved samplerate is correctly recalled by satdump-ui played locally, samplerate displayed by satdump_sdr_server used is default as displayed
(D) Samplerate sent 250000

f6bvp, 2024-01-01:
Local satdump-ui displays :+1: [12:30:00 - 01/01/2024] (I) Connecting to tcp://44.7.2.8:5656
[12:30:00 - 01/01/2024] (D) Samplerate sent 250000

satdump_sdr_server displays : 
12:30:32 - 01/01/2024] (T) Server accepted the client
[12:30:32 - 01/01/2024] (T) Waiting for client!
[12:30:32 - 01/01/2024] (I) Opening Generic RTL2832U OEM #1
[12:30:32 - 01/01/2024] (I) Source opened!
[12:30:32 - 01/01/2024] (D) Frequency sent 100000000.000000
[12:30:32 - 01/01/2024] (D) Setting source settings
[12:30:32 - 01/01/2024] (D) Bit Depth sent 16
[12:30:32 - 01/01/2024] (D) Samplerate sent 1920000
[12:30:32 - 01/01/2024] (D) Frequency sent 1544500000.000000

f6bvp, 2024-01-01:
Her eis the complete sequence 

When performing Recorder selection of remote server satdump-ui shows correct saved samplerate i.e. for example 1.92 Msps.
However it writes on local console :+1: 13:22:32 - 01/01/2024] (I) Connecting to tcp://44.7.2.8:5656
[13:22:32 - 01/01/2024] (D) Samplerate sent 250000

At the same time satdump-sdr_server displays : 
13:27:10 - 01/01/2024] (T) Server accepted the client
[13:27:10 - 01/01/2024] (T) Waiting for client!
[13:27:11 - 01/01/2024] (T) Found devices (sources) :
[13:27:11 - 01/01/2024] (T) - Generic RTL2832U OEM #0
[13:27:11 - 01/01/2024] (T) - Generic RTL2832U OEM #1
[13:27:11 - 01/01/2024] (T) Server lost client
[13:27:41 - 01/01/2024] (T) Server accepted the client
[13:27:41 - 01/01/2024] (T) Waiting for client!
[13:27:41 - 01/01/2024] (I) Opening Generic RTL2832U OEM #1
[13:27:41 - 01/01/2024] (I) Source opened!
[13:27:41 - 01/01/2024] (D) Frequency sent 100000000.000000
[13:27:41 - 01/01/2024] (D) Setting source settings
[13:27:41 - 01/01/2024] (D) Bit Depth sent 16
[13:27:41 - 01/01/2024] (D) Samplerate sent 1920000
[13:27:41 - 01/01/2024] (D) Frequency sent 1544500000.000000

So the question is, what exact sample frequency is used if we do not change it ?

f6bvp, 2024-01-01:
Answer is that the saved samplefrequency is used.
Conclusion. only message"Samplerate sent 250000"  was wrong.

JVital2013, 2024-01-01:
Yeah there's definately a bug here. From what I'm seeing, the "Samplerate sent 250000" message does not always happen, just sometimes. And, when it does happen, it triggers a few other bugs. To be investigated!

JVital2013, 2024-01-01:
This is fixed as of #c909c9ac.

The issue resides in the fact that the server will send a default samplerate if one is not selected. With the current protocol, this must happen in some situations. For example, if no default samplerate is saved on the client, the server must give it one.

Long story short - if it takes more than 50ms for the settings packet to get to the server, the server will think none is coming and will send the default to the client. The server will get the message eventually - and that's why the samplerate seems to be correct, other than the client log - but the client is now partially out of sync, since the server told it the samplerate changed. There are several side effects of this, one being the FFT/Waterfall will appear "faster" than a local SDR.

With this change, the server now updates the client with the correct samplerate, once it gets its tardy settings packet. You will still see the incorrect samplerate in the log, but it will always be followed up by the correct one.

Here is an example of what the log may look like now:

[image]

commit c909c9ac (2024-01-01): Remote SDR: Acknowledge samplerate change (fixed #576)
closed 2024-01-01 by JVital2013