SatDump/SatDump #577: (E) Error updating TLEs. Not updated

f6bvp, 2024-01-01:
(I) Loading pipelines from /usr/share/satdump/pipelines
[14:20:23 - 01/01/2024] (D) Registered pipelines :
[14:20:23 - 01/01/2024] (I) Found OpenCL Devices (0) :
[14:20:23 - 01/01/2024] (I) Loading TLEs from /home/bernard/.config/satdump/satdump_tles.txt
[14:20:23 - 01/01/2024] (I) 9139 TLEs loaded!
[14:20:23 - 01/01/2024] (I) Querying SDR Server at 44.7.2.8:5656
[14:20:24 - 01/01/2024] (D) Device File Source
[14:20:24 - 01/01/2024] (D) Device 44.7.2.8:5656 - Generic RTL2832U OEM #0
[14:20:24 - 01/01/2024] (D) Device 44.7.2.8:5656 - Generic RTL2832U OEM #1
[14:20:24 - 01/01/2024] (D) Device Generic RTL2832U OEM #0
[14:20:24 - 01/01/2024] (D) Device RTL-TCP
[14:20:24 - 01/01/2024] (D) Device SDR++ Server
[14:20:24 - 01/01/2024] (D) Device SpyServer
[14:20:24 - 01/01/2024] (D) Device UDP Source
[14:20:24 - 01/01/2024] (T) FFT Rate 8333, Samplerate 1000000, Final Size 8192, Skip 141
[14:20:24 - 01/01/2024] (I) http://celestrak.org/NORAD/elements/active.txt
[14:20:25 - 01/01/2024] (I) Failed getting TLEs. Retrying...
[14:20:25 - 01/01/2024] (W) Failed to get TLE for http://celestrak.org/NORAD/elements/active.txt
[14:20:25 - 01/01/2024] (E) Error updating TLEs. Not updated.

f6bvp, 2024-01-01:
Updating TLE from the same computer with Gpredict is OK.

JVital2013, 2024-01-01:
Please update to the latest nightly; this has been resolved. Celestrak removed legacy URLs SatDump was relying on

f6bvp, 2024-01-01:
still not updating  TLE with
(I) Starting SatDump v1.1.3-e3bf40a8 version.
Will try next nightly built.

f6bvp, 2024-01-01:
I forgot to delete satdump_cfg.json with
     "urls_to_fetch": [
 311             "http//celestrak.org/NORAD/elements/active.txt" // Most sats

JVital2013, 2024-01-01:
Yep, that's it. All we changed was the URL in that json file :-)

closed 2024-01-01 by JVital2013

WD8DUP, 2024-09-20:
On Windows 10 and using ver 1.2 here and still not updating TLEs. It seems to fail when updating one of the TLE's for 31304. Could it be that it's failing when just one TLE cannot be updated instead of skipping over.  According to Space-Track though 31304 is a valid ID.  I can see that it appears to be updating Kepler data, but when it hits this one ID, it fails.
Any ideas.

WD8DUP, 2024-09-20:
Never mind that last one.  Just installed the latest dump and it appears to be working fine.  Now just need to run through all the "bells and whistles".  
Floyd (WD8DUP/DA1VF)