SatDump/SatDump #580: SPS invalid - no NOAA APT constellation or image

gm8arv, 2024-01-03:
Just updated to the nightly Windows build, although I don't know whether that's related.  Was working fine before.  Issue now is that I'm getting no NOAA APT display ("constellation", image etc.), and a message in the status bar "SPS is invalid, Must be above 1!"  The spectrum and waterfall are correct (HF+ 192 ksps/4).  So which SPS might be wrong, and where might I find it?  I can't see anything immediately wrong in the .json file.

gm8arv, 2024-01-03:
Here's setting.json......

{
    "user_interface": {
        "play_audio": {
            "value": false
        },
        "open_viewer_post_processing": {
            "value": false
        },
        "remember_pos": {
            "value": true
        },
        "window": {
            "maximized": false,
            "x": 332,
            "y": 51,
            "xs": 1250,
            "ys": 882
        }
    },
    "satdump_general": {
        "opencl_device": {
            "platform": 1
        },
        "qth_lat": {
            "value": 55.9
        },
        "qth_lon": {
            "value": -3.2
        },
        "qth_alt": {
            "value": 180.0
        },
        "default_borders_color": {
            "value": [
                0.0,
                1.0,
                0.0
            ]
        },
        "default_shores_color": {
            "value": [
                1.0,
                1.0,
                0.0
            ]
        },
        "default_cities_color": {
            "value": [
                1.0,
                0.0,
                0.0
            ]
        },
        "default_qth_color": {
            "value": [
                1.0,
                0.0,
                1.0
            ]
        },
        "default_latlon_color": {
            "value": [
                0.0,
                0.0,
                1.0
            ]
        }
    },
[... 168 lines cut ...]
            "rotator_update_period": 1.0,
            "min_elevation": 20.0,
            "rotator_config": {
                "rotctl": {
                    "address": "127.0.0.1",
                    "port": 4533
                }
            },
            "rotator_algo": {
                "park_position": {
                    "az": 0.0,
                    "el": 0.0
                },
                "park_while_idle": false,
                "unpark_at_minus": 60.0,
                "update_period": 1.0
            }
        }
    },
    "cli": {
        "engage_autotrack": true,
        "source": "AirSpyHF",
        "start_recorder_device": true
    }
}

gm8arv, 2024-01-04:
I wonder whether the combination of 192 ksps and 4 decimation is the issue?  This results in an error recorded in the log file as the audio sample rate is 50 ksps.  Is the audio sample rate fixed, or it it /easy/ to change to 48000 ksps?  Could making the receiver bandwidth less by using a 192/4 sample rate rather than 256/4 be helpful?

"recorder_sdr_settings": {
"samplerate": 192000,
"decimation": 4

[06:52:20 - 04/01/2024] (I) Module noaa_apt_decoder
[06:52:20 - 04/01/2024] (D) Parameters :
[06:52:20 - 04/01/2024] (D)    - audio_samplerate : 50000.0
[06:52:20 - 04/01/2024] (D)    - autocrop_wedges : false
[06:52:20 - 04/01/2024] (D)    - baseband_format : "f32"
[06:52:20 - 04/01/2024] (D)    - buffer_size : 1000000
[06:52:20 - 04/01/2024] (D)    - samplerate : 48000
[06:52:20 - 04/01/2024] (D) Input SPS : 0.960000
[06:52:20 - 04/01/2024] (D) Resample : 1
[06:52:20 - 04/01/2024] (D) Samplerate : 50000.000000
[06:52:20 - 04/01/2024] (D) Dec factor : 0.960000
[06:52:20 - 04/01/2024] (D) Final SPS : 1.000000
[06:52:20 - 04/01/2024] (E) SPS is invalid. Must be above 1!

gm8arv, 2024-01-04:
Setting the Airspy HF+ Discovery to 265/4 did the trick.

Aang23, 2024-01-05:
This seemed weird to me at first, but is because the APT module requires at least 50kHz. 

As a note you don't need to decimate manually in the recorder outside of display preferences, the APT demodulator will do it internally anyway.

And while at it - happy new year!

closed 2024-01-05 by Aang23

smargle9343, 2025-06-16:
I'm having this issue while trying to demodulate meteor m2-3 lrpt. it just wont work. I'm using a wav file for the input, and I'm also using the nightly...... could that have something to do with the error?