SatDump/SatDump #581: Autostart on Reboot

ve3elb, 2024-01-04:
@JVital2013 
Hello Everyone.
I would like to have SatDump autostart on boot with user configuration setup and start receiving and recoding automatically. Maybe we can have SatDump look at a user config file on boot. 
Want to have SatDump and GOESTools start running after power failure/reboot on a remote unattended Windows PC. Any ideas. 

Thank You
Vince.

gm8arv, 2024-01-04:
Using the nightly version you can have a command like:

start "Running SatDump" /D  C:\Tools\satdump\bin\   satdump-ui.exe --start_recorder_device --engage_autotrack --source AirSpyHF

which will achieve what you want.  I have this in command script which runs on reboot.  I think this just uses the last active configuration .

ve3elb, 2024-01-04:
This looks like what am looking for. Where can I find a complete list of all the commands. 

What I need is:

Radio:
RTLSDR
Freq: 16941
Gain 49
BiasT enable

Processors:
Goes to Goestools

gm8arv, 2024-01-04:
There's some information here:

  https://www.a-centauri.com/articoli/automating-satellite-reception-with-satdump

but I'm unsure where the full list of options may be found.  The device, frequency etc. are just whatever you set in the initial setup.  I hope that's enough to get you started.

JVital2013, 2024-02-04:
Hi @ve3elb,

Unfortunately, the automated reception guide is mostly for receiving scheduled LEOs. For GEOs, you will want to start SatDump in command line mode. This will use fewer resources, and is designed more for continuous reception/automatic start.

Here's how I would do it:

1. Create a file at `%appdata%\Microsoft\Windows\Start Menu\Programs\Startup\Satdump.bat`
2. In satdump.bat, add the following lines:

```bat
@echo off
cd "C:\Program Files\SatDump\bin"
satdump live goes_hrit_tcp . --source rtlsdr --samplerate 2.4e6 --gain 49 --bias --frequency 1694.1e6
```

3. Run the bat file, or restart your computer, and CLI-based decoding of GOES to goestools will begin.

**As an added note**, running cd to your SatDump install dir is necessary, unless you install SatDump with the "Add SatDump to the system PATH" option, in which case you can omit cd'ing to the install directory.

[image]

JVital2013, 2024-03-23:
Can we provide any further assistance here?

ve3elb, 2024-03-23:
It's all good and working thank you.

closed 2024-03-23 by ve3elb