SatDump/SatDump #588: Viewer fails to display after APT decode.

Tonypattinson57, 2024-01-07:
Viewer fails to display after NOAA APT decode.
PC Win10

V1.1.3 release
[satdump-20240107T133516.log](https://github.com/SatDump/SatDump/files/13853653/satdump-20240107T133516.log)
Decoded products are created and stored in the designated locations, but are never offered in the viewer. Meteor M2-3 files decode and display without problems. No problems with V1.1.2.

JVital2013, 2024-01-07:
For now, make sure you manually set the start timestamp. It looks like this happens if you don't set the timestamp, and it can't figure out the start time from the filename.

I can reproduce the issue - even once fixed, you will want to make sure this timestamp is set to everything works properly, but we should probably make it so the pipeline does not crash here.

[image]

Tonypattinson57, 2024-01-07:
Oh, thanks for such a prompt reply. I wasn't aware that I could do 
anything with  the timestamp - it's always just worked before. I've just 
tried manually transferring the time from the filename, and it's 
produced the outputs - after a minute or two. I suppose there's a lot 
more going on than before. Does Meteor use a different system for that?

Best Regards

Tony

decode. (Issue #588)

image.png (view on web)

Aang23, 2024-01-07:
NOAA APT is Analog, and only carries the 2 image channels and some basic telemetry. There is no time information that can be used to locate each line in time (required for a lot of the processing). It needs to be provided another way.

 On the other hand, any more modern digital protocol (even DSB / HRPT from NOAA), such as METEOR LRPT carry time information internally which removes this requirement entirely.

JVital2013, 2024-01-07:
There is a slight bug here: it should be loading the data in the viewer after the pipeline finishes, even if the timestamp was not set. This worked in 1.1.2, but not 1.1.3. This is due to a config change.

**This is now fixed in the latest nightly build, which is available at https://github.com/SatDump/SatDump/releases/tag/nightly.**

Also, a quick not on "it worked fine before": If you weren't selecting the timestamp on 1.1.2, it would have worked: but only partially. I just ran a test on 1.1.2, and it will make the following files with the timestamp set to "Auto" (and will do so quite quickly):

```
APT-A.png
APT-B.png
avhrr_apt_A_individual_equalized.png
avhrr_apt_B_individual_equalized.png
raw.png
```

If you had set the start time, 1.1.2 would have made all these files

```
APT-A.png
APT-B.png
avhrr_apt_A_individual_equalized.png
avhrr_apt_B_individual_equalized.png
avhrr_apt_rgb_MCIR.png
avhrr_apt_rgb_MCIR_map.png
avhrr_apt_rgb_MCIR_Rain.png
avhrr_apt_rgb_MCIR_Rain_map.png
avhrr_apt_rgb_MSA.png
avhrr_apt_rgb_MSA_map.png
dataset.json
product.cbor
raw.png
```
It would have been much slower, since it's doing a lot more - which is why 1.1.3 seems slower when you set the timestamp. In current versions, there are even more composites than I list above. As such, it's recommended to always set the start timestamp to get the most out of your data (or, format your baseband wav filenames to match that of SDR++ or SatDump's recorder).

Thanks for coming over from Facebook to help us fix this issue, and let us know if we can be of more assistance!

Tonypattinson57, 2024-01-08:
Thanks for the explanation of the way things work, which does make 
sense. I've tried the nightly build, and that seems to work as you 
suggest, with the fast operation but no extra files. That's not a bad 
feature actually, as it's handy to be able to see if a pass has been 
worth catching. I usually use SDR Console V3.3 as my receiving software, 
and I don't think the timestamp style is configurable. Incidentally, are 
all the error messages that display during a decode necessary feature? 
They go by too fast to read, and must take up processing time to handle.

Thanks again for a quick solution to my problem.

Tony Pattinson

decode. (Issue #588)

https://github.com/SatDump/SatDump/releases/tag/nightly 
<https://github.com/SatDump/SatDump/releases/tag/nightly> .

JVital2013, 2024-01-08:
At some point we should probably do something about those errors - they're a bit "spammy". It likely won't affect speed at all, since it'll try to produce all composites it can, but we could probably clean up the interface at some point.

Also: I don't think SDR Console lets you change the filenames, but I just added support in SatDump to understand filenames from it. If you get the latest nightly, it should now auto-populate the timestamp field - as long as you select the pipeline first, then open the file.

[image]

Tonypattinson57, 2024-01-09:
Thanks for that,  Jamie. It certainly works much better, and seems to process things much quicker. It was taking up to three minutes before. My SDR Console has decided not to talk to the SDR this morning and I haven’t been able to get it back, even after a re-install, so that needs some thought. Both SDR# and SatDump still work OK with it, so it’s not the USB drivers, which has happened before.

The viewer with NOAA apt seems to still have odd holes in it that can cause one to lose control of a decoded file. What I mean is that, after decoding, one can view the individual channels etc. but attempting to use the wrong enhancement or map feature prevents further control of that particular decode. Any others that may be already decoded, or doing a fresh decode works normally. It may be case of being presented with menu items that won’t work for that data, but not failing elegantly.

I’m mentioning it here, for you information, not complaining!

Best regards

Tony

From: Jamie Vital 
Sent: Monday, January 8, 2024 9:03 PM
To: SatDump/SatDump 
Cc: Tonypattinson57 ; Author 
Subject: Re: [SatDump/SatDump] Viewer fails to display after APT decode. (Issue #588)

At some point we should probably do something about those errors - they're a bit "spammy". It likely won't affect speed at all, since it'll try to produce all composites it can, but we could probably clean up the interface at some point.

Also: I don't think SDR Console lets you change the filenames, but I just added support in SatDump to understand filenames from it. If you get the latest nightly, it should now auto-populate the timestamp field - as long as you select the pipeline first, then open the file.

image.png (view on web)

JVital2013, 2024-01-09:
Re-opening since it sounds like there may be more to discuss here.

Before we go any further: it's probably worth updating to the latest nightly again; I just made some more performance enhancements to the product processor to make it go faster, and to reduce error message spam. It will still give several error messages if you didn't set the timestamp, however. Those should probably stay, as an indicator that not all information was provided.

After you update and get a few passes under your belt: are you still experiencing any issues with selecting "the wrong composite" making the viewer lock up for that decode? If so, it would be helpful to know which composite/settings combo you're trying when it locks up.

Tonypattinson57, 2024-01-09:
Ok, I’ve just updated to the latest nightly (-093f5391).

I’ll need a bit more time to try and break it again, but something that I had noticed since your SDR-Console/SatDump  timestamp compatibility work, is that whilst the APT SDR-Console file time & date is picked up, native SatDump files aren’t anymore. They retain the last timestamp that was used. It’s taken me a while to convince myself what’s happening, but I think that’s the case. Today has been the first time that I’ve really used SatDump by itself for receiving/recording, as I’m having problems with SDR-Console. 

I’ve got a zoom meeting now, but I’ll carry on later.

Tony

From: Jamie Vital 
Sent: Tuesday, January 9, 2024 7:32 PM
To: SatDump/SatDump 
Cc: Tonypattinson57 ; Author 
Subject: Re: [SatDump/SatDump] Viewer fails to display after APT decode. (Issue #588)

Re-opening since it sounds like there may be more to discuss here.

Before we go any further: it's probably worth updating to the latest nightly again; I just made some more performance enhancements to the product processor to make it go faster, and to reduce error message spam. It will still give several error messages if you didn't set the timestamp, however. Those should probably stay, as an indicator that not all information was provided.

After you update and get a few passes under your belt: are you still experiencing any issues with selecting "the wrong composite" making the viewer lock up for that decode? If so, it would be helpful to know which composite/settings combo you're trying when it locks up.

JVital2013, 2024-01-09:
Let me know what you find with more testing, but on my PC, it is still recognizing SatDump timestamps. If I set it to apt, then select an input file from SatDump, it sets the time correctly. Not quite sure what you're running into there

Tonypattinson57, 2024-01-10:
OK, I’ve had time for another play, and I can now show that, as I said, there’s problem with the time stamp pickup on SatDump files. If I start with a fresh run of SatDump, select APT, then a old native file, the date and time will appear as current. If I first select a SDR-Console file then the timestamp picks up correctly. If I subsequently select the native file, then the previous timestamp remains, and is carried through to processing etc. I suspect that this is what caused some of my ‘holes’ and decode hang ups yesterday, as it’s trying to use wildly inaccurate time data. I did manage to exhibit that once this morning after trying to put a map outline on.

I’ll continue to exercise it and try to pin it down a bit more. I don’t have many SatDump NOAA files yet, as I don’t usually use it for receiving, but the problem is immediately apparent without running the file.

Another small point is that even when working correctly, the map positioning always overlays a little to the north, and never matches the actual land mass. Whether this is normal, I don’t know, as I’ve never had these facilities before.

Tony

From: Jamie Vital 
Sent: Tuesday, January 9, 2024 7:32 PM
To: SatDump/SatDump 
Cc: Tonypattinson57 ; Author 
Subject: Re: [SatDump/SatDump] Viewer fails to display after APT decode. (Issue #588)

Re-opening since it sounds like there may be more to discuss here.

Before we go any further: it's probably worth updating to the latest nightly again; I just made some more performance enhancements to the product processor to make it go faster, and to reduce error message spam. It will still give several error messages if you didn't set the timestamp, however. Those should probably stay, as an indicator that not all information was provided.

After you update and get a few passes under your belt: are you still experiencing any issues with selecting "the wrong composite" making the viewer lock up for that decode? If so, it would be helpful to know which composite/settings combo you're trying when it locks up.

Tonypattinson57, 2024-01-10:
Further to my last, mail. I think I’ve found how to make an APT decode hang. Nothing to do with the timestamp problem. Select any APT file and decode. When viewing, it will probably default to CH1, but it doesn’t matter which. Select Map Overlay or Lat/Long grid then Apply. An error message ‘None of the required channels are present’ briefly appears and that’s it! No further control of that decode is possible. If, on the other hand, Composite is first selected, thing seem to proceed OK. Subsequent viewing of CH1, say, will revert to a composite mode if a map outline is selected.

Tony

From: Jamie Vital 
Sent: Tuesday, January 9, 2024 7:32 PM
To: SatDump/SatDump 
Cc: Tonypattinson57 ; Author 
Subject: Re: [SatDump/SatDump] Viewer fails to display after APT decode. (Issue #588)

Re-opening since it sounds like there may be more to discuss here.

Before we go any further: it's probably worth updating to the latest nightly again; I just made some more performance enhancements to the product processor to make it go faster, and to reduce error message spam. It will still give several error messages if you didn't set the timestamp, however. Those should probably stay, as an indicator that not all information was provided.

After you update and get a few passes under your belt: are you still experiencing any issues with selecting "the wrong composite" making the viewer lock up for that decode? If so, it would be helpful to know which composite/settings combo you're trying when it locks up.

JVital2013, 2024-01-10:
I'm doing some testing: does the map overlay/subsequent lock-up in the viewer only happen when the start timestamp was not set on initial decode? I can reproduce it in a few situations then, which we should fix. But I can't reproduce it when the data was decoded with a start timestamp.

As for the satdump file name issue, can you share the exact filename SatDump saved, including extension, that does not set the start timestamp when you load it in offline processing?

Tonypattinson57, 2024-01-10:
I’m somewhat embarrassed to say that I can’t make it fail now. If I wait long enough after pressing ‘Apply’, the image reappears with the map and I can carry on. I tried it so many times this morning and it failed every time. 

I do have another in the same area which crashes SatDump. ‘Night Fire detection’ crashes it every time for me at the moment.

This is the Satdump filename that doesn’t set the timestamp correctly:
E:\WXSAT DATA\Baseband data\2024-01-09_11-54-38-145_1792000SPS_137912500Hz.wav

Tony

From: Jamie Vital 
Sent: Wednesday, January 10, 2024 9:06 PM
To: SatDump/SatDump 
Cc: Tonypattinson57 ; Author 
Subject: Re: [SatDump/SatDump] Viewer fails to display after APT decode. (Issue #588)

I'm doing some testing: does the map overlay/subsequent lock-up in the viewer only happen when the start timestamp was not set on initial decode? I can reproduce it in a few situations then, which we should fix. But I can't reproduce it when the data was decoded with a start timestamp.

As for the satdump file name issue, can you share the exact filename SatDump saved, including extension, that does not set the start timestamp when you load it in offline processing?

JVital2013, 2024-01-10:
Thanks for all the info! I believe I can reproduce all these issues now in specific situations. We'll work on it and will get back to you.

JVital2013, 2024-01-15:
Hi @Tonypattinson57,

I believe these bugs are fixed as of the latest nightly. Could you try updating again, and see if you have any more issues?

- **Nighttime fire Detection Crash:** Fixed
- **Map overlay lockups:** Fixed - caused by errors in some composites when information necessary for the map overlay was missing
- **Filename issue:** Fixed - caused by turning on milliseconds in filename in SatDump settings. The milliseconds field won't be imported at this time, but the rest of it will fill in correctly

Tonypattinson57, 2024-01-15:
Hi Jamie,

so far so good. I haven’t managed to actually break it yet. 

I note that not first selecting APT after, say, a Meteor file, will cause the date info to revert to current, as will selecting the file before the type. I know it’s not how it should be operated, but if you don’t notice it, the decode will have the wrong time info.

I miss having the 221 composite option on NOAA decodes. Was there a good reason for removing it?

Regards

Tony

From: Jamie Vital 
Sent: Monday, January 15, 2024 1:57 AM
To: SatDump/SatDump 
Cc: Tonypattinson57 ; Mention 
Subject: Re: [SatDump/SatDump] Viewer fails to display after APT decode. (Issue #588)

Hi @Tonypattinson57,

I believe these bugs are fixed as of the latest nightly. Could you try updating again, and see if you have any more issues?

  a.. Nighttime fire Detection Crash: Fixed 
  b.. Map overlay lockups: Fixed - caused by errors in some composites when information necessary for the map overlay was missing 
  c.. Filename issue: Fixed - caused by turning on milliseconds in filename in SatDump settings. The milliseconds field won't be imported at this time, but the rest of it will fill in correctly

JVital2013, 2024-01-15:
The 221 composite is gone from APT since no APT transmission ever has channel 2 and 1 at the same time. It still works with NOAA HRPT.

As for the timestamp not being set if the file is picked first, then the pipeline: I do see what you're saying here. We _might_ be able to do something about that!

Tonypattinson57, 2024-01-15:
Ah, so that’s it. I’m not clear on what channels are available, as it seems to vary a bit between them. Whatever it did, when it was there, gave an RGB approximation of daylight terrain. 

Are we now saying that this is closed, or do you want to carry on a bit? I’m happy enough to test things, but you probably have better things to get on with.

Tony

From: Jamie Vital 
Sent: Monday, January 15, 2024 5:50 PM
To: SatDump/SatDump 
Cc: Tonypattinson57 ; Mention 
Subject: Re: [SatDump/SatDump] Viewer fails to display after APT decode. (Issue #588)

The 221 composite is gone from APT since no APT transmission ever has channel 2 and 1 at the same time. It still works with NOAA HRPT.

As for the timestamp not being set if the file is picked first, then the pipeline: I do see what you're saying here. We might be able to do something about that!

JVital2013, 2024-01-16:
The next nightly build will now honor the timestamp/sample rate from the input file, even when switching pipelines. This should fix things when selecting the pipeline after selecting the input file.

At this point I think I'll call this completed since it sounds like everything's working as expected for you. If you have any other issues, feel free to reach back out!

closed 2024-01-16 by JVital2013

Tonypattinson57, 2024-01-17:
Thanks for all your work. It seems to be much more reliable now.
Is there any information as to how the Stratosat-TK1 decoder should work? I’ve recorded several good passes, and the decoder provides a *.frm file of data, but I’ve no idea what to do with it now.

Tony

From: Jamie Vital 
Sent: Tuesday, January 16, 2024 8:29 PM
To: SatDump/SatDump 
Cc: Tonypattinson57 ; Mention 
Subject: Re: [SatDump/SatDump] Viewer fails to display after APT decode. (Issue #588)

The next nightly build will now honor the timestamp/sample rate from the input file, even when switching pipelines. This should fix things when selecting the pipeline after selecting the input file.

At this point I think I'll call this completed since it sounds like everything's working as expected for you. If you have any other issues, feel free to reach back out!

JVital2013, 2024-01-17:
The Stratosat-TK1 decoder is very much still a work-in-process. Right now, you essentially need a perfect decode to get an image out of it - if you're just getting a frm, then SatDump failed to save the image - likely due to missing/corrupt frames. There's more work to do here.