SatDump/SatDump #589: Error compiling on Mac Book Pro M1 Max / Ventura

RubbaCippi, 2024-01-07:
Hi, 
I followed the compile instruction on https://github.com/SatDump/SatDump#macos
When I reach the build and install ing, specifically the line
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON -DNNG_ENABLE_TLS=ON ..
I get a first error, 
rob@M1Max build % cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON -DNNG_ENABLE_TLS=ON ..
CMake Error at CMakeLists.txt:28 (cmake_minimum_required):
  CMake 3.13 or higher is required.  You are running version 3.3.2

I can fix editing the line in the source file.

Then running again I get various errors:

rob@M1Max build % cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON -DNNG_ENABLE_TLS=ON ..
-- The C compiler identification is AppleClang 15.0.0.15000100
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring for NNG version 1.6.0
You have called ADD_LIBRARY for library nng without any source files. This typically indicates a problem with your CMakeLists.txt file
You have called ADD_LIBRARY for library nng_testing without any source files. This typically indicates a problem with your CMakeLists.txt file
-- Performing Test NNG_HIDDEN_VISIBILITY
-- Performing Test NNG_HIDDEN_VISIBILITY - Success
-- Looking for strlcpy
-- Looking for strlcpy - found
-- Looking for strnlen
-- Looking for strnlen - found
-- Looking for strcasecmp
-- Looking for strcasecmp - found
-- Looking for strncasecmp
-- Looking for strncasecmp - found
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
CMake Error at cmake/NNGHelpers.cmake:54 (target_link_libraries):
  Attempt to add link library "Threads::Threads" to target "nng" which is not
  built in this directory.
Call Stack (most recent call first):
  src/platform/posix/CMakeLists.txt:17 (nng_link_libraries)

-- Looking for lockf
-- Looking for lockf - found
-- Looking for flock
-- Looking for flock - found
-- Looking for getrandom
-- Looking for getrandom - not found
-- Looking for arc4random_buf
-- Looking for arc4random_buf - found
-- Looking for clock_gettime in rt
-- Looking for clock_gettime in rt - not found
-- Looking for sem_wait in pthread
-- Looking for sem_wait in pthread - found
CMake Error at cmake/NNGHelpers.cmake:151 (target_link_libraries):
  Attempt to add link library "pthread" to target "nng" which is not built in
  this directory.
Call Stack (most recent call first):
  src/platform/posix/CMakeLists.txt:33 (nng_check_lib)

CMake Error at cmake/NNGHelpers.cmake:152 (target_link_libraries):
  Attempt to add link library "pthread" to target "nng_testing" which is not
  built in this directory.
Call Stack (most recent call first):
  src/platform/posix/CMakeLists.txt:33 (nng_check_lib)

-- Looking for pthread_atfork in pthread
-- Looking for pthread_atfork in pthread - found
CMake Error at cmake/NNGHelpers.cmake:151 (target_link_libraries):
  Attempt to add link library "pthread" to target "nng" which is not built in
  this directory.
Call Stack (most recent call first):
  src/platform/posix/CMakeLists.txt:34 (nng_check_lib)

CMake Error at cmake/NNGHelpers.cmake:152 (target_link_libraries):
  Attempt to add link library "pthread" to target "nng_testing" which is not
  built in this directory.
Call Stack (most recent call first):
  src/platform/posix/CMakeLists.txt:34 (nng_check_lib)

-- Looking for pthread_set_name_np in pthread
-- Looking for pthread_set_name_np in pthread - not found
-- Looking for pthread_setname_np in pthread
-- Looking for pthread_setname_np in pthread - found
CMake Error at cmake/NNGHelpers.cmake:151 (target_link_libraries):
  Attempt to add link library "pthread" to target "nng" which is not built in
  this directory.
Call Stack (most recent call first):
  src/platform/posix/CMakeLists.txt:36 (nng_check_lib)

CMake Error at cmake/NNGHelpers.cmake:152 (target_link_libraries):
  Attempt to add link library "pthread" to target "nng_testing" which is not
  built in this directory.
Call Stack (most recent call first):
  src/platform/posix/CMakeLists.txt:36 (nng_check_lib)

-- Looking for gethostbyname in nsl
-- Looking for gethostbyname in nsl - not found
-- Looking for socket in socket
-- Looking for socket in socket - not found
-- Looking for __atomic_load_1 in atomic
-- Looking for __atomic_load_1 in atomic - not found
-- Looking for AF_UNIX
-- Looking for AF_UNIX - found
-- Looking for backtrace_symbols_fd
-- Looking for backtrace_symbols_fd - found
-- Performing Test NNG_HAVE_MSG_CONTROL
-- Performing Test NNG_HAVE_MSG_CONTROL - Success
-- Looking for eventfd
-- Looking for eventfd - not found
-- Looking for kqueue
-- Looking for kqueue - found
-- Looking for port_create
-- Looking for port_create - not found
-- Looking for epoll_create
-- Looking for epoll_create - not found
-- Looking for epoll_create1
-- Looking for epoll_create1 - not found
-- Looking for getpeereid
-- Looking for getpeereid - found
-- Looking for SO_PEERCRED
-- Looking for SO_PEERCRED - not found
-- Performing Test NNG_HAVE_SOCKPEERCRED
-- Performing Test NNG_HAVE_SOCKPEERCRED - Failed
-- Looking for LOCAL_PEERCRED
-- Looking for LOCAL_PEERCRED - found
-- Looking for LOCAL_PEERPID
-- Looking for LOCAL_PEERPID - found
-- Looking for getpeerucred
-- Looking for getpeerucred - not found
-- Looking for atomic_flag_test_and_set
-- Looking for atomic_flag_test_and_set - found
CMake Warning at src/supplemental/tls/mbedtls/CMakeLists.txt:11 (message):
  

          ************************************************************
          Linking against Mbed TLS changes license terms (Apache 2.0).
          Consult a lawyer and the license files for details.
          ************************************************************

-- Found mbedTLS: /usr/local/lib/libmbedtls.dylib (found version "3.5.1") 
CMake Error at cmake/NNGHelpers.cmake:54 (target_link_libraries):
  Attempt to add link library "/usr/local/lib/libmbedtls.dylib" to target
  "nng" which is not built in this directory.
Call Stack (most recent call first):
  src/supplemental/tls/mbedtls/CMakeLists.txt:27 (nng_link_libraries)

CMake Error at cmake/NNGHelpers.cmake:54 (target_link_libraries):
  Attempt to add link library "/usr/local/lib/libmbedx509.dylib" to target
  "nng" which is not built in this directory.
Call Stack (most recent call first):
  src/supplemental/tls/mbedtls/CMakeLists.txt:27 (nng_link_libraries)

CMake Error at cmake/NNGHelpers.cmake:54 (target_link_libraries):
  Attempt to add link library "/usr/local/lib/libmbedcrypto.dylib" to target
  "nng" which is not built in this directory.
Call Stack (most recent call first):
  src/supplemental/tls/mbedtls/CMakeLists.txt:27 (nng_link_libraries)

CMake Error at src/CMakeLists.txt:51 (install):
  install TARGETS given target "nng" which does not exist in this directory.

-- Looking for nl_langinfo
-- Looking for nl_langinfo - found
-- Test zt disabled (unconfigured)
-- The CXX compiler identification is AppleClang 15.0.0.15000100
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring incomplete, errors occurred!
See also "/Users/rob/nng/build/CMakeFiles/CMakeOutput.log".
See also "/Users/rob/nng/build/CMakeFiles/CMakeError.log".
rob@M1Max build % 

-- Configuring incomplete, errors occurred!

Look as some libraries was missing.

**Hardware (SDR/PC/OS)**

MacBook Pro 14" M1 max, macOS Ventura 13.5.2, Xcode updated,

**Version (Eg, 1.0.0, CI Build 171)**

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

JVital2013, 2024-01-07:
Your cmake version (3.3.2) is quite out of date. Make sure you have the latest version installed with `brew install cmake`, then try again.

After updating with brew, you can run `cmake --version` to  verify you have the latest version. Right now, the version Brew has available is 3.28.1

RubbaCippi, 2024-01-07:
Hi Jamie,
thank you for the reply and support.
Look as I have cmake 3.28.1 but the version return me 3.3.2
And the version of brew is 4.2.2
I don’t know how to force to run the 3.28.1 instead of 3.3

Ciao
Roberto

==================================================================
***@***.*** ~ % cmake --version
cmake version 3.3.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
***@***.*** ~ % brew install cmake
==> Downloading https://formulae.brew.sh/api/formula.jws.json
##################################################################################################################################################################################################################### 100.0%
==> Downloading https://formulae.brew.sh/api/cask.jws.json
##################################################################################################################################################################################################################### 100.0%
Warning: Treating cmake as a formula. For the cask, use homebrew/cask/cmake
Warning: cmake 3.28.1 is already installed and up-to-date.
To reinstall 3.28.1, run:
  brew reinstall cmake
***@***.*** ~ % cmake --version
cmake version 3.3.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
***@***.*** ~ %

JVital2013, 2024-01-07:
The brew version looks right, but not cmake. It sounds like the link to cmake might be pointing to the wrong copy. Try running this in terminal to see if it helps

```zsh
brew uninstall --force cmake
brew install --force cmake
```

Take note of any errors brew gives you. Then, run `cmake --version` to see if it's updated. If it's not, run `ls -l $(which cmake)` to see if it's linked to the version Brew installed

RubbaCippi, 2024-01-07:
after running the two brew command, I get

***@***.*** bin % ls -l $(which cmake)   
-rwxr-xr-x  1 root  wheel  4382604  5 Ott  2015 /opt/local/bin/cmake

***@***.*** ~ % cmake --version           
cmake version 3.3.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
***@***.*** ~ % 

So, the link point to the old, 3.3, cmake.
I don’t know where the 3.28.1 has been installed and thus to create a symbol link to it.

The new seem to be here:

***@***.*** bin % pwd

***@***.*** bin % ls -lsa
total 105808
    0 drwxr-xr-x  6 rob  admin       192 14 Dic 16:47 .
    0 drwxr-xr-x  9 rob  admin       288  7 Gen 21:25 ..
25304 -r-xr-xr-x  1 rob  admin  12954488 14 Dic 16:47 ccmake
26312 -r-xr-xr-x  1 rob  admin  13468992 14 Dic 16:47 cmake
26152 -r-xr-xr-x  1 rob  admin  13388664 14 Dic 16:47 cpack
28040 -r-xr-xr-x  1 rob  admin  14355240 14 Dic 16:47 ctest
***@***.*** bin % 

Any help will be appreciated, as for the previous help

JVital2013, 2024-01-07:
`brew info cmake` will show where it's installed, but it might be best not to mess with the link right away. Try this first:

```zsh
brew cleanup
brew doctor
```

The first command should clean up old versions, and the second command should tell you if there are any errors.

One thing I'm wondering is: has `/opt/homebrew/bin` been added to your system PATH? At one point setting this up was manual on Silicon macOS - it might still be.

RubbaCippi, 2024-01-08:
Thank you for the continuous support.
I tried also to add /opt/homebrew/bin to the path with no success.
I plan, as I find some free time, to remove and reinstall from scratch all the brew.

Ciao
Roberto

JVital2013, 2024-01-08:
Sounds like a good plan; I suspect your issue lies there

RubbaCippi, 2024-01-10:
Hi Jamie,

I removed and reinstalled Homebrew with no juice: 
cmake — version returned me 3.3.2

So I searched internet on how update cmake, and found a app on https://cmake.org/download/
Running this app updated silently make, so also in terminal I got 3.28.1

I then tried to compile SatDump.
At the end was found two errors, but the .dmg contain an app which run.
Next time I find some time I’ll try to recompile SatDump and see how to fix the errors.

Thank you for you support
Roberto

JVital2013, 2024-02-11:
Hello,

Since you were able to get the app to build, I'm going to close this issue. FYI, we now have native Silicon apps available in the nightlies that you can use for full performance on Silicon hardware, without building it yourself.

closed 2024-02-11 by JVital2013

RubbaCippi, 2024-02-11:
Thank you!
It was very kind from you to keep me updated.

Ciao
Roberto

RubbaCippi, 2024-02-11:
Hi Jamie,
I downloaded the compiled SatDump for silicon, and it crash on opening.
I attach the crash log.

I just updated to the last upgrade of Ventura, 13.6.4 and it is running on a MacBook Pro 14” with M1

If you need more info, don’t esitate to ask!

Ciao
Roberto

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               satdump-ui [19181]
Path:                  /Applications/SatDump.app/Contents/MacOS/satdump-ui
Identifier:            com.altillimity.satdump
Version:               1.1.4 (???)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               502

Date/Time:             2024-02-11 17:58:09.3946 +0100
OS Version:            macOS 13.6.4 (22G513)
Report Version:        12
Anonymous UUID:        877F10E4-E84B-5EB9-D8BA-4BB495496AE4

Time Awake Since Boot: 6600 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   satdump-ui [19181]

Application Specific Information:
abort() called

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x18ca10704 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x18ca47c28 pthread_kill + 288
2   libsystem_c.dylib             	       0x18c955ae8 abort + 180
3   libc++abi.dylib               	       0x18ca00b84 abort_message + 132
4   libc++abi.dylib               	       0x18c9f03b4 demangling_terminate_handler() + 320
5   libobjc.A.dylib               	       0x18c6c6e68 _objc_terminate() + 160
6   libc++abi.dylib               	       0x18c9fff48 std::__terminate(void (*)()) + 16
7   libc++abi.dylib               	       0x18ca02d34 __cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 36
8   libc++abi.dylib               	       0x18ca02ce0 __cxa_throw + 140
9   libsatdump_core.dylib         	       0x103b5e998 bool nlohmann::json_abi_v3_11_2::detail::json_sax_dom_parser<nlohmann::json_abi_v3_11_2::basic_json<nlohmann::json_abi_v3_11_2::ordered_map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::json_abi_v3_11_2::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>, void>>::parse_error<nlohmann::json_abi_v3_11_2::detail::parse_error>(unsigned long, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, nlohmann::json_abi_v3_11_2::detail::parse_error const&) + 84
10  libsatdump_core.dylib         	       0x103ee71b0 bool nlohmann::json_abi_v3_11_2::detail::parser<nlohmann::json_abi_v3_11_2::basic_json<nlohmann::json_abi_v3_11_2::ordered_map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::json_abi_v3_11_2::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>, void>, nlohmann::json_abi_v3_11_2::detail::input_stream_adapter>::sax_parse_internal<nlohmann::json_abi_v3_11_2::detail::json_sax_dom_parser<nlohmann::json_abi_v3_11_2::basic_json<nlohmann::json_abi_v3_11_2::ordered_map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::json_abi_v3_11_2::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>, void>>>(nlohmann::json_abi_v3_11_2::detail::json_sax_dom_parser<nlohmann::json_abi_v3_11_2::basic_json<nlohmann::json_abi_v3_11_2::ordered_map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::json_abi_v3_11_2::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>, void>>*) + 2720
11  libsatdump_core.dylib         	       0x103ee2930 nlohmann::json_abi_v3_11_2::detail::parser<nlohmann::json_abi_v3_11_2::basic_json<nlohmann::json_abi_v3_11_2::ordered_map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::json_abi_v3_11_2::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>, void>, nlohmann::json_abi_v3_11_2::detail::input_stream_adapter>::parse(bool, nlohmann::json_abi_v3_11_2::basic_json<nlohmann::json_abi_v3_11_2::ordered_map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::json_abi_v3_11_2::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>, void>&) + 132
12  libsatdump_core.dylib         	       0x103ee2790 nlohmann::json_abi_v3_11_2::operator>>(std::__1::basic_istream<char, std::__1::char_traits<char>>&, nlohmann::json_abi_v3_11_2::basic_json<nlohmann::json_abi_v3_11_2::ordered_map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::alloc [... cut ...]
Display: SONY TV, 1920 x 1080 (1080p FHD - Full High Definition), MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: May 27 2023 01:34:59 version 20.96.31.0.8.7.148 FWID 01-126f2c4c
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: JMS56x Series
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

RubbaCippi, 2024-02-11:
I am again here with a bad new:
also the previous version I compiled now crash on startup.
The only difference is the version of the OS, 13.6.4 instead of 13.6.3…

I’ll try to recompile and keep you informed.

Ciao
Roberto

RubbaCippi, 2024-02-11:
again…

the crash dump for the version I copied write:

Library not loaded: @rpath/libsatdump_core.dylib
Referenced from: <4ED7D27C-8090-30DC-A9F9-A1E96E435944> /Applications/SatDump.app/Contents/MacOS/satdump-ui
Reason: tried: '/usr/local/lib/libsatdump_core.dylib' (no such file),

I copied the libsatdump_core.dylib from the app to '/usr/local/lib/ and it worked again.
I don’t know how and why the libsatdump_core.dylan was removed.

Don’t seem to be the same problem as the one of the nightly build.

Ciao
Roberto

RubbaCippi, 2024-02-11:
> There was an unrelated bug that caused a crash on all platforms - I just fixed it. Download the latest silicon nightly and try again

ok, just downloaded and work!

Ciao
Roberto

JVital2013, 2024-02-11:
Great, glad to hear it!