SatDump/SatDump #595: satdump-ui  does not disconnect and stop streaming after recording a satellite pass

f6bvp, 2024-01-12:
When some satellite pass are selected in menu Schedule and Config for Live and record the application satdump-ui starts at AOS but does not stop streaming and does no disconnect from remote satdump_sdr_server.

f6bvp, 2024-01-12:
at LOS of satellite

JVital2013, 2024-01-12:
This isn't really a bug - it's simply not designed to shut down the SDR when the pass completes at this point. That can be changed, though. 

What are your thoughts on this @Aang23? I've gotten a few requests for this personally. Perhaps it should be an option in the scheduler config?

f6bvp, 2024-01-13:
I don't know exactly how this is managed. Connection followed by SDR start and streaming ?
If SDR is started at AOS , SDR should be stopped at LOS by disconnecting.
Letting the streaming open after LOS is very bandpass consuming on the network when working remotely.

Aang23, 2024-01-16:
Indeed at the moment it is not designed to shutdown the SDR at all. It starts it mainly to avoid user error. Though yes, as there have been many requests to do so it'd make sense to implement a simple checkbox to change this behavior.

Both make sense to be kept as depending on the SDR, keeping it running can increase reliability. Take something such as SDRPlays, it is not uncommon for them to fail to stop/restart (despite being reliable when streaming) which would be an issue on this kind of setup.

JVital2013, 2024-01-17:
This has been implemented with the latest nightly - you can find the setting in the Tracking Configuration.

[image]

closed 2024-01-17 by JVital2013