SatDump/SatDump #596: SatDump hangs whole system

Vitaspiros, 2024-01-13:
**Description of the issue**
When I open satdump-ui, whole system becomes unresponsive. Sometimes the mouse stops moving and for some reason the Caps Lock key light starts flashing on and off, sometimes it doesn't hang immediately, but I can't kill satdump-ui using `pkill -9` and cannot restart the computer. 
**Hardware (SDR/PC/OS)**
Arch Linux, Hyprland (Wayland)
Laptop with AMD Ryzen 5 5500U
AMDGPU open source for graphics driver.

**Version (Eg, 1.0.0, CI Build 171)**
Built from source, commit 21a11f86e545c3b7aeca75d6519830d04745a52e
**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
```
[09:17:54 - 13/01/2024] (E) Usage : satdump-ui [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[09:17:54 - 13/01/2024] (E) Extra options (examples. Any parameter used in modules can be used here) :
[09:17:54 - 13/01/2024] (E)  --samplerate [baseband_samplerate] --baseband_format [f32/i16/i8/w8] --dc_block --iq_swap
[09:17:54 - 13/01/2024] (D) Starting with OpenGL 4.6 (Compatibility Profile) Mesa 23.3.3-arch1.1
[09:17:54 - 13/01/2024] (I)    _____       __  ____                      
[09:17:54 - 13/01/2024] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[09:17:54 - 13/01/2024] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[09:17:54 - 13/01/2024] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[09:17:54 - 13/01/2024] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[09:17:54 - 13/01/2024] (I)                                     /_/      
[09:17:54 - 13/01/2024] (I) Starting SatDump v1.1.4-21a11f86
[09:17:54 - 13/01/2024] (I) 
[09:17:54 - 13/01/2024] (I) Loading config /usr/share/satdump/satdump_cfg.json
[09:17:54 - 13/01/2024] (I) Loading user settings /home/spiros/.config/satdump/settings.json
[09:17:54 - 13/01/2024] (I) Loading plugins from /usr/lib/satdump/plugins
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libspyserver_support.so...
[09:17:54 - 13/01/2024] (T) Plugin spyserver_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/liborbcomm_support.so...
[09:17:54 - 13/01/2024] (T) Plugin orbcomm_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libsdrpp_server_support.so...
[09:17:54 - 13/01/2024] (T) Plugin spyserver_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libplutosdr_sdr_support.so...
[09:17:54 - 13/01/2024] (T) Plugin plutosdr_sdr_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libstereo_support.so...
[09:17:54 - 13/01/2024] (T) Plugin stereo_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libelektro_arktika_support.so...
[09:17:54 - 13/01/2024] (T) Plugin elektro_arktika_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libsimd_sse41.so...
[09:17:54 - 13/01/2024] (T) Plugin simd_sse41 loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libanalog_support.so...
[09:17:54 - 13/01/2024] (T) Plugin analog_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libumka_support.so...
[09:17:54 - 13/01/2024] (T) Plugin umka_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libinmarsat_support.so...
[09:17:54 - 13/01/2024] (T) Plugin inmarsat_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libhinode_support.so...
[09:17:54 - 13/01/2024] (T) Plugin hinode_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libhimawari_support.so...
[09:17:54 - 13/01/2024] (T) Plugin himawari_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libbladerf_sdr_support.so...
[09:17:54 - 13/01/2024] (T) Plugin bladerf_sdr_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libeos_support.so...
[09:17:54 - 13/01/2024] (T) Plugin eos_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libairspy_sdr_support.so...
[09:17:54 - 13/01/2024] (T) Plugin airspy_sdr_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libmeteor_support.so...
[09:17:54 - 13/01/2024] (T) Plugin meteor_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/librtlsdr_sdr_support.so...
[09:17:54 - 13/01/2024] (T) Plugin rtlsdr_sdr_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libdscovr_support.so...
[09:17:54 - 13/01/2024] (T) Plugin dscovr_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libgeonetcast_support.so...
[09:17:54 - 13/01/2024] (T) Plugin geonetcast_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libfengyun4_support.so...
[09:17:54 - 13/01/2024] (T) Plugin fenyun4_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libspacex_support.so...
[09:17:54 - 13/01/2024] (T) Plugin spacex_support loaded!
[09:17:54 - 13/01/2024] (T) Loading plugin /usr/lib/satdump/plugins/libproba_support.so...
[09:17:54 - 13/01/2024] (T) Plugin proba_support loaded!
[... 447 lines cut ...]
[09:17:54 - 13/01/2024] (D)  - sentinel6_dump
[09:17:54 - 13/01/2024] (D)  - slim_tlm_link
[09:17:54 - 13/01/2024] (D)  - smap_s_link
[09:17:54 - 13/01/2024] (D)  - smos_tlm
[09:17:54 - 13/01/2024] (D)  - starship_tlm
[09:17:54 - 13/01/2024] (D)  - stereo_hr
[09:17:54 - 13/01/2024] (D)  - stereo_lr
[09:17:54 - 13/01/2024] (D)  - stratosat_tk1_uhf
[09:17:54 - 13/01/2024] (D)  - npp_hrd
[09:17:54 - 13/01/2024] (D)  - swot_tlm
[09:17:54 - 13/01/2024] (D)  - syracuse3b_tlm
[09:17:54 - 13/01/2024] (D)  - terra_db
[09:17:54 - 13/01/2024] (D)  - test_custom
[09:17:54 - 13/01/2024] (D)  - tiahne1_tlm
[09:17:54 - 13/01/2024] (D)  - tianwen1_link
[09:17:54 - 13/01/2024] (D)  - timed_dump
[09:17:54 - 13/01/2024] (D)  - tubin_x_dump
[09:17:54 - 13/01/2024] (D)  - umka_1_dump
[09:17:54 - 13/01/2024] (D)  - viasat3_tlm_12200
[09:17:54 - 13/01/2024] (D)  - viasat3_tlm
[09:17:54 - 13/01/2024] (D)  - youthsat_dump
[09:17:54 - 13/01/2024] (D)  - yunhai_ahrpt
```
**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
The system was unresponsive, so I couldn't take a screenshot.

JVital2013, 2024-01-15:
A flashing caps lock key means a kernel panic in Linux. This can be a large number of things, likely a driver issue in this case.

We have had several reports of issues with the AMDGPU driver, including myself. It seems the best solution at this time is to uninstall the driver, and re-install just the OpenCL components to see if they work.

This driver also breaks SDR++ on my system. It might be worth trying that to see if that program runs correctly.

Vitaspiros, 2024-01-16:
Thank you for responding.

> This driver also breaks SDR++ on my system. It might be worth trying that to see if that program runs correctly.

Yes, SDR++ is my recording program and it works perfectly.

Also, at first, SatDump worked normally on my system. Some driver update may have caused it to malfunction. 

> We have had several reports of issues with the AMDGPU driver, including myself. It seems the best solution at this time is to uninstall the driver, and re-install just the OpenCL components to see if they work.

I seem to can't just uninstall the driver because too many things are dependent on it:
```
spiros@xontros-gatos:~/Downloads
> sudo pacman -R mesa
[sudo] password for spiros: 
checking dependencies...
error: failed to prepare transaction (could not satisfy dependencies)
:: removing mesa breaks dependency 'mesa' required by gst-plugins-base-libs
:: removing mesa breaks dependency 'opengl-driver' required by hyprland
:: removing mesa breaks dependency 'mesa' required by lib32-mesa
:: removing mesa breaks dependency 'mesa' required by libglvnd
:: removing mesa breaks dependency 'opengl-driver' required by libglvnd
:: removing mesa breaks dependency 'mesa' required by qemu-ui-opengl
:: removing mesa breaks dependency 'mesa' required by qemu-vhost-user-gpu
:: removing mesa breaks dependency 'mesa' required by qt5-base
:: removing mesa breaks dependency 'mesa' required by qt6-base
:: removing mesa breaks dependency 'mesa' required by rocm-opencl-runtime
:: removing mesa breaks dependency 'mesa' required by virglrenderer
:: removing mesa breaks dependency 'mesa' required by webkit2gtk
:: removing mesa breaks dependency 'mesa' required by webkit2gtk-4.1
:: removing mesa breaks dependency 'opengl-driver' required by wlroots
:: removing mesa breaks dependency 'mesa' required by xorg-server-devel
:: removing mesa breaks dependency 'mesa' required by xorg-xwayland
```

JVital2013, 2024-01-16:
Ah, I forgot how Arch does the amdgpu driver. Removing mesa is a bad idea so we'll need to try something else.

If SDR++ works, it might be the OpenCL component of the driver causing the problem with SatDump. Try rebuilding SatDump from source, but add `-DBUILD_OPENCL=OFF` to the cmake parameters to see if that helps.

Vitaspiros, 2024-01-26:
Sorry for the late response.
I tried that and now SatDump works. I am closing this issue now.

closed 2024-01-26 by Vitaspiros