SatDump/SatDump #597: Save record only when satellite is above selected minimum elevation

f6bvp, 2024-01-13:
In Schedule and config menu there is presently an option about satellite minimum elevation.
This selects satellite passes with at least requested minimum elevation. Good.
However saved files on disk record the whole pass from AOS to LOS, not only when elevation is above minimum.
When using omnidirectional antenna such as a patch the signal SNR is not optimum due to low gain and satellite slant range. 
In order to reduce size of recorded files without interesting signal could it be possible to only save part of the pass over selected minimal elevation ?
Here is a complete record displayed with Audacity, showing the part with stronger signal.
[image]

f6bvp, 2024-01-13:
For post processing I am using Audacity to remove manually unwanted parts.
[image]

f6bvp, 2024-01-13:
By the way. 
As can be seen wav16 files are interpreted as 32 bits by audacity. Who is wrong ?

JohnJohnston852, 2024-05-12:
I too would like to see this feature implemented. If I've set the minimum elevation then presumably I've set it for a reason, such as the signal being too weak or obstructed below that elevation. It doesn't make sense to use up a ton of disk space recording just noise for large sections of the passes.