SatDump/SatDump #600: Elektro L4 Misaligned Images with low SNR

rtlsdrblog, 2024-01-15:
I'm getting about 3-4 dB SNR on Elektro-L4, and images are coming in, but getting a perfect image is quite rare. Often the images are corrupted with misaligned sections.

This is using a 65cm discovery dish with linear feed. I realize the dish size might too small for Elektro, and also the feed is linear instead of circular, hence the SNR is probably too low resulting in some errors in the decoding. From what I've seen from others more than 6 dB is probably required for perfect images.

But is there any possible enhancements to the way SatDump stitches the segments together? For example in this image almost everything is there apart from some missing data in the lower right, and this appears to have caused a full misaligned segment. If the misaligned section was at least aligned the image would be usable.

[image]
[image]

On the color images there is also often corruption with what looks like the color components not aligning.

[image]

Also while I'm here, has anyone else noticed that while Elektro is circularly polarized, it also has a slight linear polarization to it? Optimizing the linear feed angle results in a difference of about 2 dB.

Aang23, 2024-01-16:
Hello!

I think it would be better if you could record and send over a full baseband. This kind of SNR should be plenty for something good so it may be some tuning being required. The shifts you see would be very complicated to fix though unfortunately due to the structure of the data transmitted.

Yes, it seems like all ELEKTROs (and also Arktika) have somewhat of a linear component as well. Not the first time I see it.

Aang23, 2024-01-16:
Also : Do make sure to be using the latest version, a bug fixed not so long ago could have been part of the issue.

rtlsdrblog, 2024-01-16:
Thanks, I'll try and get a baseband to you soon. Ironically I just recorded a baseband a few minutes ago during the last scheduled transmit, but I got an average of 4.1dB and all perfect images this time after making a few tweaks to the dish position. So the threshold for perfect images is maybe only 4 dB. Previously I was about 3.5 - 3.8 dB.

> Also : Do make sure to be using the latest version, a bug fixed not so long ago could have been part of the issue.

Curious, what commit # is that? Will check if I'm already running past that commit or not.

Aang23, 2024-01-16:
On paper, even under 1db is enough for a clean decode if you don't have noise or such. I decode L3 at such SNRs before.

It was this commit : https://github.com/SatDump/SatDump/commit/1094de355758f2bf36d47cfd49327eab61041e3d

rtlsdrblog, 2024-01-17:
Ok so this might actually be a problem with live decoding only. 

I just did a live decode and also recorded baseband during the last activation. During the live decoding I saw the error "LRIT CRC Invalid, but file can be recovered" a few times and whenever that happened the image preview showed a corrupted misaligned segment like what I uploaded in the original post.

But then when running the .s8 recording through the offline decoder, I end up with perfect images.

I'm doing all this on an Orange Pi 5 running Ubuntu. During decoding I checked htop, and CPU usage was stable about 50-60%.

rtlsdrblog, 2024-01-17:
Here's a link to a recording. https://mega.nz/file/5HMgAa4B#alotO1FBouDZNFEW0X8Lik8kBWFn5ggmnaYmMqTKY5c

It was done with an Airspy Mini at 6 MSPS, with decimation of 4. I recorded it as .s8 by mistake, but the offline output is almost the same as what I got with live decoding. 

During live decoding the average SNR was about 4.2 dB and I got a few "LRIT CRC invalid, but file can be recovered" messages, and whenever those messages occurred I could see that the latest segment in the image preview was misaligned.

rtlsdrblog, 2024-01-19:
Here are two more recordings with the Airspy recorded at 16-bit and in ziq format. SNR is at at about 4.2 dB.

https://mega.nz/file/JDlzgS7Z#sN5bIZy46EqZCJhyz7mRnMG6Pwp9mV3mTuDaUrqFYk0
https://mega.nz/file/cecklZbC#MqAZdU6fHCCmuT0UuFRCB8MPne903SASC4A9PWRM8zw

I notice that with these recordings if I reduce the pll_bw to 0.01 I get much better images but a few glitches still remain however. 

But reducing the pll_bw doesn't help with the first recording in the last reply. Possibly because I recorded the first file in .s8 which seems to result in the recording getting lower SNR at only 3.2 dB. But if 1dB is supposed to be enough, I don't know why it's creating glitched images.

rtlsdrblog, 2024-01-22:
On further investigation this issue might be the same as the originally reported deframer sync issue in https://github.com/SatDump/SatDump/issues/287

I believe it to be either a processing power issue, or an issue with the code on ARM. But the Orange Pi 5 I'm using should have plenty of grunt, and htop shows CPU usage on all cores only hovering around 50% during reception and decoding. The Orange Pi 5 I'm using is also fan cooled, so it never throttles. The Orange Pi 5 is also fine with GOES HRIT, HRPT, and GK2A, FengYun LRIT.

I tested by changing the CPU governor to performance (initially ondemand was selected), and decimating by 5 from 2.4 MSPS with the RTL-SDR. This results in sometimes perfect images at 3.2dB, but sometimes with still one of two images glitched. The more I decimate (with the limit at 5 for 2.4 MSPS on the RTL-SDR), the more likely I am to get perfect images I think. 

I'm not sure if it's dropping samples or what, but the deframer definitely loses sync, and the ziq files recorded on the Orange Pi 5 don't produce any different results if I run them on an x64 machine. I haven't tested recording data with the Elektro-L decoder not running yet however.

Also I just did a test with rtl_tcp on the Orange Pi 5 sending data to Satdump running on a powerful x64 Linux machine and got perfect images at 3.2 dB without any decimation used.

rtlsdrblog, 2024-01-25:
Okay after a bit of testing with a Elektro L4 LRIT signal played back via a HackRF I believe I've found a workaround for this issue at least for the Orange Pi 5.

The fix is to use rtl_tcp as the source on the same OPI5 machine instead of the built in RTL-SDR source, and use 2.56MSPS with a decimation greater than three (usually I use the max decimation of 5 for this LRIT signal). This results in perfect images, even at SNRs less than three.

If I use rtl_tcp alone without decimation there are some small brief deframer sync losses resulting in glitches, but at least the entire image isn't corrupted like when using the native RTL-SDR source in SatDump. Oddly using 1.024 MSPS as the sample rate instead of decimating causes a lot of deframer sync losses too. Also I think I was wrong about the CPU governor, that doesn't really do anything.

It seems that some stage in the pipeline is getting clogged up and dropping data even thought it seems like there is sufficient CPU power left. Taking the RTL-SDR source out of the SatDump thread helps a bit, as does reducing the amount of data the decoder has to process with decimation. Just a guess but It might be related to RS error correction because deframer sync loss occurs when the RS numbers turn orange, and this of course happens more often with low SNR signals that need more error correction done.

I still want to test what @pluggerlockett said in https://github.com/SatDump/SatDump/issues/287, about this problem not occurring on v0.39.

rtlsdrblog, 2024-01-25:
Just tested and confirmed that v0.39 decodes Elektro LRIT perfectly, so I think this definitely must be a regression that came with the 1.0.0 architecture.

pluggerlockett, 2024-01-29:
Hey @rtlsdrblog, just thought I'd chime in and give you a bit of an update from my perspective. As you rightly called out I had massive issues with Raspberry Pi's (3A/3B) running Satdump v1.x and live decoding L3 LRIT via the command line like I historically did with v0.39 on the same hardware. I was never able to get this to work reliably and I can't help but blame either the architecture or lack of grunt.
I've since migrated to a Wo-We Mini PC with AMD Excavator A9-9400. I now run Satdump-ui via KDE on Debian 12 on that box on my station, it's so much more stable and provides clean decodes more often when compared to my old Pi/v0.39 station. They're currently $80USD with free shipping.

rtlsdrblog, 2024-01-30:
@pluggerlockett thanks for chiming in and confirming the issue. Interesting that even a Pi 3 worked for you on v0.0.39. I was expecting a Pi 4 to be the minimum. But that does give me hope that there is something that can be fixed in the code, and it's not just that these ARM devices are actually underpowered.

In my case I would prefer to stick with ARM devices because of their much lower power usage compared to x64 devices. I'm powering everything over PoE so there's only a limited power budget as I'm also powering a small dish rotator from PoE too. As I mentioned, all the other pipelines (GOES HRIT, GK-2A, AHRPT) appear to be working fine in my setup, apart from Elektro (possibly FengYun-4 is also displaying a similar issue but more rarely).

Not sure if @Aang23 has any ideas on what changed between V0.0.39 and V1.0.0 that could cause this?

rtlsdrblog, 2024-01-31:
I just submitted a PR https://github.com/SatDump/SatDump/pull/616 which I believe fixes this issue, at least from my tests on the Orange Pi 5.

First decrease the MAX_SPS value back to what it was in earlier versions. This appears to increase the amount of decimation used on the signal.

Secondly, increase the rtlsdr_read_async buffer back to the higher fixed value it was in older versions. The newer versions appear to set a buffer size based on sample rate, but it seems to be too small to not lose samples.

rtlsdrblog, 2024-02-01:
Closing as fixed in https://github.com/SatDump/SatDump/commit/99044be156e1e9d5fe3fc78102f74703fbe32f86

closed 2024-02-01 by rtlsdrblog