SatDump/SatDump #603: Himawaricast decoder - compression error message Bzip2

eswnl, 2024-01-18:
Encountered the following error when outputting Himawaricast images. It affects the visible channel because of higher resolution possibly. Here the TS file was input into Satdump directly.

[image]

Link to the TS file:
https://drive.google.com/file/d/1-Pbtb54LQq7KQRDufss6s1moI0Iug7rU/view?usp=drive_link

Oh, also I think the following needs to be added for demodulation:
"cadu": {
                "s2_udp_cadu_extractor": {
                    "bb_size": 38688,
                    "pid": 1001
                }
            },

eswnl, 2024-02-04:
Hi Jamie,

I changed the link setting to general access. You can try below:
https://drive.google.com/file/d/1-Pbtb54LQq7KQRDufss6s1moI0Iug7rU/view?usp=sharing

There are 2 separate issues, one is demodulation and the other is decoding.

JVital2013, 2024-02-04:
Thanks! I did a bit of investigating - in this example, the file that is failing is IMG_DK01IR3_202401171411_007.bz2, so not the VIS channel this time. It's failing because the CRC check is coming up invalid within the bzip file. TSReader also complains about a continuity error in the ts. It looks to be bad data to me, but perhaps you're thinking otherwise?

Extracted bz2 segment: [IMG_DK01IR3_202401171411_007.bz2.zip](https://github.com/SatDump/SatDump/files/14158677/IMG_DK01IR3_202401171411_007.bz2.zip)

[image]

eswnl, 2024-02-05:
I think I know what's happened.
The Himawari data set includes the visible LRIT (2200x2200) and HRIT (11000x11000). The problem is both come under the same file name and the HRIT is being overwritten by the smaller file LRIT as you can see in the explorer.
[image]

Below is the TS stream:
https://drive.google.com/file/d/1UxsC_dHN4ht4GE-1ILauU2gMazwL4VXl/view?usp=sharing

It's a quick fix.

JVital2013, 2024-02-05:
Ah yes, good catch! Although, it looks like the two different resolution images do have different names - they are off by one minute. SatDump was just saving some of them under the wrong name, resulting in an overwrite in some situations.

I now get both vis channels:

[image]

Interestingly, it looks like this had an effect on some of the IR channels too, but on that side, the HRIT and LRIT images with the same channel name look quite different:

[image]

Try the latest nightly and let me know what you think. I still see the bz2 error, but those appear to be data stream errors.

eswnl, 2024-02-07:
I just tested it and it works fine. As for the other error, it could be a break in the transport stream when I stopped the recording. 
I kind of wish satdump could have TS input socket to perform a live decode from an external source as I don't think satdump's DVB-S2 demodulator does so well with weaker signals.

closed 2024-02-15 by JVital2013