SatDump/SatDump #605: The actual frequency of the setting at LO Frequency offset

Aybarnur, 2024-01-19:
The signal receiving from the insat-3dr satellite in Windows 8.1.
I use a  in C-band LNB-converter with LO=5950 MHz.
The received frequency is 4781.5 MHz, in the converter it's subtracted from the LO frequency: 5950 - 4781.5 = 1169.5 MHz.

With unadjusted tuning scale, reception is successful at the frequency 1169.5 MHz.
[image]

If I set LO Frequency offset 5950 in the LO field, then the scale shows the actual reception frequency of 4781.5 MHz.
[image]

However, there is no reception and I find a strange value in the log:
[07:11:21 - 19/01/2024] (T) Using PlutoSDR IP Address 192.168.64.61
[07:11:21 - 19/01/2024] (D) Set PlutoSDR samplerate to 8000000
[07:11:22 - 19/01/2024] (D) Set PlutoSDR frequency to 485532704
[07:11:22 - 19/01/2024] (D) Set PlutoSDR gain to 33, mode manual
[07:11:22 - 19/01/2024] (T) FFT Rate 66666, Samplerate 8000000, Final Size 65536, Skip 1130
[07:11:36 - 19/01/2024] (I) Waiting for the thread...
[07:11:36 - 19/01/2024] (I) Thread stopped
[07:11:36 - 19/01/2024] (I) Saving user config at ./config/settings.json

[image]

JVital2013, 2024-02-04:
There is a slight mistake in your math: 5950 - 4781.5 = 116**8**.5, not 116**9**.5. Minor point, but I thought I'd bring it up.

If you set the LO field to 5950, then you want to put the **actual** frequency of the satellite in the frequency field. SatDump will do the math for you. The UI will show the actual frequency you're receiving over the air, and the SDR is tuned to the correctly shifted frequency, as shown here:

[image]

Aybarnur, 2024-02-06:
Thank you, it really was my mistake in calculations. Now I'm using the correct value and everything is working fine.
I would also like to interesting if there is any news about the processing of data and get images from this satellite?

[image]

JVital2013, 2024-02-06:
Glad to hear it's working for you! Unfortunately there's no interesting news on that satellite as of now; we will update #436 once there is progress

closed 2024-02-06 by JVital2013