SatDump/SatDump #606: armadilo on Fedora 39

burgeda, 2024-01-19:
Recently Fedora updated armadilo from version 10 to version 12.

Now satdump-ui fails to launch as it seems to be hard coded to use libarmdilo.so.10

Adding a symbolic link 

ln -s /usr/lib64/libarmadilo.so.12 /usr/lib64/libarmadilo.so.10

fixes it ... for now.

Aang23, 2024-01-19:
If you built manually you should re-run CMake rather to have it find the new Armadillo.

burgeda, 2024-01-19:
I did.

---- On Fri, 19 Jan 2024 15:55:36 -0500 Aang23 ***@***.***> wrote ---

If you built manually you should re-run CMake rather to have it find the new Armadillo.

burgeda, 2024-01-19:
I should have been more clear.

I completely rebuilt from source.

cloned from git and ran the entire rebuild and recompile process.

---- On Fri, 19 Jan 2024 15:55:36 -0500 Aang23 ***@***.***> wrote ---

If you built manually you should re-run CMake rather to have it find the new Armadillo.

Aang23, 2024-01-19:
Oh. That's very odd. And it still picked Armadillo 10!?

Aang23, 2024-01-20:
I'll give this a go in a VM....

burgeda, 2024-01-20:
Hi

I just did the same thing with a F39 VM that hadn't been updated in a couple of weeks.

Updated the OS and all packages, rebuilt satdump from github and it worked. I compared the version to my physical box and they are both running SatDump v1.1.4-c6d45321

The VM one works fine but the physical one complains about needing armadilo.so.10.  I've already rebuilt satdump 4 times on it so, I'm going to blow the physical one back to bare metal and start over with it again.

Thanks for looking into this.

---- On Sat, 20 Jan 2024 06:04:37 -0500 Aang23 ***@***.***> wrote ---

I'll give this a go in a VM....

Aang23, 2024-01-23:
I've tried on my end as well, and it picks up the right version as expected.

closed 2024-02-16 by Aang23