SatDump/SatDump #610: Run satdump_sdr_server without a config file

burgeda, 2024-01-25:
**What would you want to be added to satdump?**
A clear and concise description of what the idea/problem is.

**Describe the feature you'd like to have**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Here you can for provide more information about what you'd like to see as the end result. Feel free to also post images if you have ideas regarding the UI

burgeda, 2024-01-25:
Right now when running satdump_sdr_server on a remote headless box such as a pi the satdump_sdr_server refuses to run without a /home/satdump/.config/satdump directory.
It doesn't actually write to that directory but it MUST be there for it to run.
When running this way it's best from a security point of view to run as a user with no login.

Can satdump_sdr_server be modified to not require this empty directory or perhaps have an option to specify where to find it?

JVital2013, 2024-01-26:
I just pushed an update that should let SatDump SDR server run without this directory; please update and let us know if it works as expected for you!

closed 2024-02-14 by Aang23