SatDump/SatDump #611: Satdump crashes when processing Meteor M2

JhoGraessl, 2024-01-25:
**Description of the issue**
Satdump closes/crashes without any error message when decoding Meteor m2 lrpt. both in offline processing and in recording
**Hardware (SDR/PC/OS)**
RTL-SDR V4, Huawei Matebook D14 2020 (Ryzen 7), KDE Neon (Ubuntu based) (also happens on windows 10)
**Version (Eg, 1.0.0, CI Build 171)**
both with 1.1.3 and now 1.1.4-2509aa7
**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
[satdump-20240125T231042.log](https://github.com/SatDump/SatDump/files/14058186/satdump-20240125T231042.log)
**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

JVital2013, 2024-01-26:
It appears you have selected the wrong pipeline. The filename indicates you are trying to decode is METEOR M2-3, but you selected the Meteor M2 pipeline. Try it with "METEOR M2-x LRPT 72k"

[image]

JhoGraessl, 2024-01-26:
> It appears you have selected the wrong pipeline. The filename indicates you are trying to decode is METEOR M2-3, but you selected the Meteor M2 pipeline. Try it with "METEOR M2-x LRPT 72k"
> 
> [image]
> 

Yes, that will be right. None the less it crashes when I want to live record.

closed 2024-01-26 by JhoGraessl

Aang23, 2024-01-26:
It also appears you were using a baseband as "soft". That's not right, you need to select the input level to be baseband.