SatDump/SatDump #612: [json.exception.type_error.302] type must be string, but is null

f6bvp, 2024-01-26:
I often get this core dumped after git pull and make a new version of satdump-ui : 

(D)  - youthsat_dump
[19:05:58 - 26/01/2024] (D)  - yunhai_ahrpt
[19:05:58 - 26/01/2024] (I) Found OpenCL Devices (0) :
[19:05:58 - 26/01/2024] (I) Loading TLEs from /home/bernard/.config/satdump/satdump_tles.txt
[19:05:58 - 26/01/2024] (I) 9218 TLEs loaded!
terminate called after throwing an instance of 'nlohmann::json_abi_v3_11_2::detail::type_error'
  what():  [json.exception.type_error.302] type must be string, but is null
Abandon (core dumped)

f6bvp, 2024-01-27:
I am launching satdump-ui from my home directory where I lef a copy of satdump_cfg.json.
After removing satdump_cfg.json from home directory thereis no more error.
When this happens, if it could help I would be pleased to compile the project with -ggdb2 flag and run it under gdb.
However I don't know where to put this flag ?

JVital2013, 2024-02-04:
Glad to hear that worked for you!

SatDump can be compiled so that it works in GDB by replacing "Release" with "Debug" in the cmake command, like so:

```bash
cmake -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_PREFIX=/usr ..
```

This works in most situations by using the -g flag internally, but if you specifically need -ggdb2, then re-run the initial cmake command like this: 

```bash
cmake -DCMAKE_CXX_FLAGS_DEBUG="-ggdb2" -DCMAKE_C_FLAGS_DEBUG="-ggdb2" -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_PREFIX=/usr ..
```

closed 2024-02-04 by JVital2013

f6bvp, 2024-02-06:
The error [json.exception.type_error.302] is still there after rebuilding version with a cron.daily script :
#!/bin/sh
#
# Mise à jour de SatDump
#
cd /home/f6bvp/satdump
git pull
#
cd build
#
sleep 1
make clean
#
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr ..
sleep 1
make
#
sudo rm -fr /usr/bin/satdump*
sudo rm -fr /usr/share/satdump
sudo rm -fr /usr/lib/satdump
sudo rm -fr /usr/include/satdump
#
sleep 1
sudo make install
#

Rhen runing satdump-ui :
...
08:54:37 - 06/02/2024] (D)  - umka_1_dump
[08:54:37 - 06/02/2024] (D)  - viasat3_tlm_12200
[08:54:37 - 06/02/2024] (D)  - viasat3_tlm
[08:54:37 - 06/02/2024] (D)  - youthsat_dump
[08:54:37 - 06/02/2024] (D)  - yunhai_ahrpt
[08:54:37 - 06/02/2024] (I) Found OpenCL Devices (0) :
[08:54:37 - 06/02/2024] (I) Loading TLEs from /home/f6bvp/.config/satdump/satdump_tles.txt
[08:54:37 - 06/02/2024] (I) 9169 TLEs loaded!
terminate called after throwing an instance of 'nlohmann::json_abi_v3_11_2::detail::type_error'
  what():  [json.exception.type_error.302] type must be string, but is null
Abandon (core dumped)

f6bvp, 2024-02-06:
I will try to trace the error

f6bvp, 2024-02-06:
The null string must be inside file satdump_cfg.json that is in my home directory.
When I rename the file satdump_cfg.json.old the error is no more present.

Aang23, 2024-02-06:
That is expected behavior, to let you override the default satdump_cfg.json with a local one if you are running in a dev environment.