SatDump/SatDump #617: Problem with Satdump Noaa and Meteor improvements

jackouille83, 2024-01-31:
**Description of the issue**
Layers do not match the received image
**Hardware (SDR/PC/OS)**
PC Windows 10 -Key RTLSDR v3

**Version (Eg, 1.0.0, CI Build 171)**
Version 1.13 and 1.14

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
[image]
[image]
[image]

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

Aang23, 2024-01-31:
Is your PC's time right? And how are you decoding the data?

jackouille83, 2024-01-31:
Yes, my pc is right time

I have the same problem on another pc, i tested Satdump V1.13 and V1.14, i have same problem on both.

I reinstall soft but same problem.

what is the best solution to reset the program? I deleted the satdump entries in the registry with regedit, but still this shift in the layers, shift which is not always the same?

Aang23, 2024-01-31:
How are you decoding the data?

jackouille83, 2024-01-31:
Either I let satdump do it automatically or via Viewer, open dataset.json or open product .cbor and then I choose an mcir or msa or other option. I am in France in the Var, my time zone is set to Grenwitch.

Aang23, 2024-01-31:
Hmmm, this is interesting. Your timezone is set to Greenwitch, as well as region? Though to be fair it does compute satellite positions, so this is weird. Can you provide the full log file?

By the way : Hi from France as well! :-) I'm west of Paris.

jackouille83, 2024-02-01:
Re, est-il possible de communiquer en francais ?

merci

jackouille83, 2024-02-01:
Rebonjour,

Les logs en pièce jointe et un exemple de ce matin

Par contre il y a aussi un exemple réussi mais aucun changement de ma part dans les reglages

C’est un mystère !!

 

Merci 

 

2015-049A
1 40899U 15049A   24031.48257431  .00018541  00000+0  57495-3 0  9994
2 40899  97.4760  45.2720 0012297  84.3245 275.9399 15.33392262463498

2022-019A
1 51824U 22019A   24031.45868132  .00011055  00000+0  51746-3 0  9998
2 51824  97.4481 105.2878 0016719   7.1468 353.0002 15.19783230106342

2022-019B
1 51825U 22019B   24031.49660597  .00009340  00000+0  45764-3 0  9995
2 51825  97.4462 104.8100 0015531   5.9895 354.1522 15.18261189106315

2022-019D
1 51827U 22019D   24031.45833878  .00008701  00000+0  42473-3 0  9991
2 51827  97.4502 105.2157 0014001   6.3398 353.8010 15.18434509106320

2022-019E
1 51828U 22019E   24031.46170713  .00015000  00000+0  63886-3 0  9996
2 51828  97.4500 106.8339 0013293   1.3969 358.7302 15.22990963106435

2022-019F
1 51829U 22019F   24031.49301476  .00034822  00000+0  10854-2 0  9991
2 51829  97.4546 110.5477 0009360 354.5980   5.5160 15.33127881106653

2022-019G
1 51830U 22019G   24031.49475384  .00035192  00000+0  10976-2 0  9999
2 51830  97.4450 109.8441 0010962 355.6304   4.4840 15.33089864106654

2022-019J
1 51832U 22019J   24031.49882404  .00007186  00000+0  40325-3 0  9990
2 51832  97.4493 103.5929 0010208  21.5253 338.6406 15.13604507106228

2022-019K
1 51833U 22019K   24031.48305389  .00018961  00000+0  73045-3 0  9990
2 51833  97.4471 108.1025 0011511 358.8695   1.2515 15.26331451106534

2022-019N
1 51836U 22019N   24031.44421827  .00031590  00000+0  99891-3 0  9995
2 51836  97.4526 110.3873 0010940 350.1002  10.0023 15.32674100106660

2022-019P
1 51837U 22019P   24031.47147619  .00034941  00000+0  10474-2 0  9995
2 51837  97.4451 110.4905 0008615 359.6027   0.5207 15.34374034106700

2022-019U
1 51842U 22019U   24031.46257161  .00048950  00000+0  83170-3 0  9993
2 51842  97.5156 130.0479 0038462 241.0873 118.6518 15.50456573107624

2022-019V
1 51843U 22019V   24031.47658285  .00007672  00000+0  37932-3 0  9998
2 51843  97.4509 105.8154 0013795  18.6990 341.4747 15.18030128106353

2022-019X
1 51845U 22019X   24031.50351348  .00017110  00000+0  68985-3 0  9994
2 51845  97.4472 107.9392 0012025   4.3594 355.7746 15.24825976106434

[... 1617 lines cut ...]
1 43602U 18067A   24028.50299802  .00000055  00000+0  00000+0 0  9996
2 43602  54.2417 210.5845 0008173  31.3261 328.7140  1.86231700 36901

BEIDOU-3 M13
1 43622U 18072A   24031.28151842 -.00000025  00000+0  00000+0 0  9992
2 43622  55.8828  91.4832 0007553 286.0712  73.9150  1.86231254 36497

BEIDOU-3 M14
1 43623U 18072B   24030.61002145 -.00000023  00000+0  00000+0 0  9998
2 43623  55.8843  91.4987 0004882 305.8155  54.2097  1.86231241 36475

BEIDOU-3 M15
1 43648U 18078B   24028.21770866 -.00000094  00000+0  00000+0 0  9994
2 43648  54.8114 330.3093 0004914   8.3960 351.6207  1.86232239 35968

BEIDOU-3 M16
1 43647U 18078A   24031.31030154 -.00000086  00000+0  00000+0 0  9998
2 43647  54.8134 330.1994 0008029   6.4987 353.5269  1.86232334 36007

BEIDOU-3 M17
1 43706U 18093A   24031.05090043  .00000086  00000+0  00000+0 0  9994
2 43706  54.2353 211.7678 0006256 294.0431  65.8853  1.86228457 35352

BEIDOU-3 M18
1 43707U 18093

jackouille83, 2024-02-02:
Rebonjour,

J’ai trouvé d’où venait le problème, j’ai oublié de préciser le sat dans Tracking configuration mes sats étaient tous notés comme Noaa 15, d’où le décalage des masques.

Tout est ok maintenant

Merci de m’avoir répondu

JVital2013, 2024-02-04:
Yep, setting the satellite in the pipeline settings is needed at this time. Thanks for getting back to us that you figured it out!

closed 2024-02-04 by JVital2013

jackouille83, 2024-02-04:
It's me who thanks you for this great work you do with Satdump, I'm 72 years old and sometimes it takes me time to understand...........