SatDump/SatDump #618: Visualizing NOAA DSB Telemetry Data

var-IO-reeks, 2024-01-31:
Hello Satdump Team,

could you add an option to the viewer tab to visualize the telemetry data transmitted by the NOAA satellites to show data like the Scan motor Current, the subcomms and the solar array data in some kind of graphs directly within satdump?

I think a feature like that could give some additional interesting insight into the behavior and status of the NOAAs.

mpentler, 2024-03-09:
I also wondered about this Aang. Is good suggestion!

thomasdouwes, 2025-06-08:
Hi, I'm working on a total rewrite of the TIP telemetry decoder right now and this will be added, hopefully I will have it done for the 2.0 update

thomasdouwes, 2025-08-07:
This was merged to verywip in #1012. The solar array data and subcomms are not visualised, but the AVHRR Scan motor Current as well as most other AVHRR telemetry is available and now correct. There is no documentation on the solar array data so I left it out. Unfortunately with the remaining POES satellites getting decommissioned soon I don't know how useful it will be